espressif/esp-serial-flasher #4: Source code Licence

JHedgesBartec, 2020-06-05:
Hello,

I know this isn't an issue but found it may be the best way to ask what licence this code falls under?

I am researching for a method of updatng a ESP32-WROOM-32 acting as a slave from a STM32 MCU.

Regards,
Jack

MaValki, 2020-06-05:
Hi,
It is distributed under Apache 2.0 license. Sorry for missing LICENCE file, it's been added to master branch.

BTW, I am currently working on port for STM32. However, for the sake of simplicity `read()` `write()` functions will be implemented with polling `HAL_UART_Transmit/Receive`. In case, such approach does not fulfill your requirements and you decide to write port on your own, don't hesitate to open another issue, if you encounter any problems.

JHedgesBartec, 2020-06-05:
Hi Martin,

Thanks for the clarification on that. Do you have a timescale for the port? We would also be looking at polled transmit/receive rather than interrupt driven.

Thanks,
Jack

MaValki, 2020-06-08:
Hi, I expect it to be pushed to github in 2 weeks.