espressif/esp-serial-flasher #5: DEBUG: Error: COMMAND_FAILED    [E][FLASHER.ino:60] flash_bin(): Flash start operation failed.

Hoel, 2020-06-11:
I ran the project as is in the arduino ecosystem which has the lastest ESP32 core (updated yesterday), and while it connects successfully to the target it fails to even start flashing.

The wiring has been double checked and is conform to the defined ports, which is confirmed by the fact it does connect correctly.
I added debug statements in loader_port_serial_write and loader_port_serial_read just to see what was sent and received, some datas are exchanged, log below.

Is this example even functional?
It shouldnt change anything that it runs in arduino ecosystem since the underlaying SDK is the same, in the arduino setup function i only call app_main(); and nothing else.

```
DEBUG: Error: COMMAND_FAILED
[E][FLASHER.ino:60] flash_bin(): Flash start operation failed.
```

```
W C0
W 0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555
W C0
R 65
R 74
R 73
R 20
R 4A
R 75
R 6E
R 20
R 20
R 38
R 20
R 32
R 30
R 31
R 36
R 20
R 30
R 30
R 3A
R 32
R 32
R 3A
R 35
R 37
R 0D
R 0A
R 0D
R 0A
R 72
R 73
R 74
R 3A
R 30
R 78
R 31
R 20
R 28
R 50
R 4F
R 57
R 45
R 52
R 4F
R 4E
R 5F
R 52
R 45
R 53
R 45
R 54
R 29
R 2C
R 62
R 6F
R 6F
R 74
R 3A
R 30
R 78
R 32
R 20
R 28
R 44
R 4F
R 57
R 4E
R 4C
R 4F
R 41
R 44
R 5F
R 42
R 4F
R 4F
R 54
R 28
R 55
R 41
R 52
R 54
R 30
R 2F
R 55
R 41
R 52
R 54
R 31
R 2F
R 53
R 44
R 49
R 4F
R 5F
R 52
R 45
R 49
R 5F
R 46
R 45
R 4F
R 5F
R 56
R 32
R 29
R 29
R 0D
R 0A
R 77
R 61
R 69
R 74
R 69
R 6E
R 67
R 20
R 66
R 6F
R 72
R 20
R 64
R 6F
R 77
R 6E
R 6C
R 6F
R 61
R 64
R 0D
R 0A
R 0A
W C0
W 0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555
W C0
R 00
W C0
W 0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555
W C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
W C0
W 000D0800000000000000000000000000
W C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
R C0
R 01
R 08
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
R C0
R 01
R 0D
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
[I][FLASHER.ino:42] flash_bin(): Connected to target
W C0
W 000F08000000000000100E0000000000
W C0
R C0
R 01
R 0F
R 04
R 00
R 12
R 20
R 55
R 55
R 00
R 00
R 00
R 00
R C0
W C0
W 0002100000000000003000000C0000000004000000D03F00
W C0
R D0
R C0
R 01
R 02
R 04
R 00
R 12
R 20
R 55
R 55
R 01
R 06
R 00
R 00
R C0
DEBUG: Error: COMMAND_FAILED
[E][FLASHER.ino:60] flash_bin(): Flash start operation failed.
```

Hoel, 2020-06-12:
UPDATE :
After more testing i found the source of the problem (but not how to solve it).
Apparently the reson for the error is the offset. 
In my case i need to flash a binary of length 0x3000 at offset 0x3FD000.
If i set the offset to 0x10000 everything works fine.
If i set the offset to 0x3FD000 then i get **Flash start operation failed**.
So for some reason the function **esp_loader_flash_start** cannot handle another offset than 0x10000 or at least not it cannot handle 0x3FD000.
Please let me know how i could fix this issue because i really have to flash at 0x3FD000, flashing at 0x10000 is not an option here.

Hoel, 2020-06-12:
UPDATE : 
So i found the problem, it was related to the ROM which assume the flash to be 2MB, and so anything after this limit is considered invalid.
I found the PR from someone else adressing this issue by adding a command to set flash parameters before **esp_loader_flash_start** https://github.com/espressif/esp-serial-flasher/pull/3/files
which fixed the problem.

closed 2020-06-12 by Hoel

MaValki, 2020-06-15:
Hi Hoel,
Thank you for contributing with detailed description of this issue. As you mentioned above, it has already been reported, and I have implemented functionality that queries flash size and adjusts it in ROM accordingly. I expect those changes to be merged into master in one week.