espressif/esp-serial-flasher #6: How to get the MAC address of the target

Hoel, 2020-06-12:
in my use case i need to generate an UUID from the target MAC address, how can i get it from the ROM with a command?

igrr, 2020-06-12:
Probably you could read the MAC address registers using the `esp_loader_read_register` function.

On the ESP32, the registers are these ones:
https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/components/soc/soc/esp32/include/soc/efuse_reg.h#L65-L79

Hoel, 2020-06-12:
A quick test with 
```
void get_id(){

  uint32_t val=0;
  esp_loader_read_register(EFUSE_BLK0_RDATA1_REG, &val);  
  printf("%08x\n",val);
  esp_loader_read_register(EFUSE_BLK0_RDATA2_REG, &val); 
  printf("%08x\n",val);
}
```
after target connection successful returns : 
```
00000000
00000000
```
what am i doing wrong?

igrr, 2020-06-12:
Could you check, does esp_load_read_register return ESP_LOADER_SUCCESS?

Hoel, 2020-06-12:
Returns error 2
```
Connected to target
error : 2
00000000
error : 2
00000000
Start programming
Finished programming
Flash verified
```

Hoel, 2020-06-12:
i moved the call after the baudrate is changed to 921600, now there is no error but the returned values seems a little suspicious.

```
uint32_t read_ifuse(uint32_t pos){
  
   uint32_t val=0;
   esp_loader_error_t err = esp_loader_read_register(EFUSE_RD_REG_BASE + (4 * pos), &val);  
   if (err != ESP_LOADER_SUCCESS) {
     printf("error : %d\n",err);
   }   
   return val;     
}
void get_id(){
  
  printf("%08x\n",read_ifuse(2)); 
  printf("%08x\n",read_ifuse(1));
}
```

```
Connected to target
bad00bad
bad00bad
Start programming
Finished programming
Flash verified
```

Hoel, 2020-06-12:
another attempt with the OTP registers you previously pointed to : 

```
void get_id(){

  uint32_t val=0;
  esp_loader_error_t err;
  err = esp_loader_read_register(ESP_OTP_MAC0, &val);
  if (err != ESP_LOADER_SUCCESS) { printf("error : %d\n",err); }
  printf("%08x\n",val); 
  err = esp_loader_read_register(ESP_OTP_MAC1, &val);
  if (err != ESP_LOADER_SUCCESS) { printf("error : %d\n",err); }
  printf("%08x\n",val); 
  err = esp_loader_read_register(ESP_OTP_MAC3, &val);
  if (err != ESP_LOADER_SUCCESS) { printf("error : %d\n",err); }
  printf("%08x\n",val);   
  //printf("%08x\n",read_ifuse(2)); 
  //printf("%08x\n",read_ifuse(1));
}
```

```
Connected to target
00000000
00000000
000008ff
Start programming
Finished programming
Flash verified
```

Hoel, 2020-06-12:
sadely i have to reopen the issue because i dont get the correct MAC.
The mac reported by esptoolPy is : 
`MAC: 24:0a:c4:04:6a:a0`
But the MAC i get when using read_ifuse(2) read_ifuse(1) is : 
`MAC 84:0d:8e:11:ad:a4`
There must be another register to get the same MAC as esptool.py .

closed 2020-06-12 by Hoel