espressif/esp-serial-flasher #8: Factory image programming

JHedgesBartec, 2020-06-24:
Hello,

What binary does this library take to program the ESP32? Looking at your example it states that the SPIFFS partition is programmed and not the factory partition.

Ideally I would be looking at taking the application binary, ignoring bootloader and partitions.bin as they won't change, and programming that directly ESP32 as the factory image. I believe this is what the ESP Flash Download Tool does, is this possible via the ESP32 UART interface?
I have the capability to put the module into the boot mode and access to the serial programming header from a STM32 MCU.

Many thanks,
Jack

MaValki, 2020-06-24:
Hi Jack,

`esp-serial-flasher` is capable of updating any binary to any address (given that address is valid and binary fits into FLASH memory). It is simplified C implementation of `esptool` (ESP Flash Download Tool) communicating through UART. SPIFFS is merely another partition used for storing binary to be flashed. 
Indeed, updating partition such as bootloader, partition tabble, NVS etc. in not mandatory. Updating application only is sufficient in many cases.
With regard to STM32, simple port along side with example will be merged into master soon. 

Regards,
Martin

JHedgesBartec, 2020-06-24:
Hi Martin,

Thankyou for clarifying that, I think I mis-read the documentation in the examples repository regarding the use of SPIFFS.
I presume the only time programming the application alone is when both the partition table and the bootloader remain unchanged?
Thats good news, I know you previously said you were hoping to have it merged to master by now. I will likely need to be using this in the coming days so if it is not available I will take a look and contribute where possible, though I am not very experienced in all things ESP,  do have experience with STM32F1xx and STM32F4xx though.

Thanks,
Jack

MaValki, 2020-06-25:
Correct, there is no need to update bootloader or partiotion table when remained unchanged.
Sorry for taking so long to merge. I have just merged it into `master`. Please take a look and let me know, if you find anything unclear.

closed 2020-07-21 by MaValki