espressif/esp-serial-flasher #9: `esp_loader_connect` fails in `detect_chip`

campbell-seer, 2020-07-21:
I'm currently integrating this library into one of our boards and have noticed that the ESP32 sends 8 packets in response to `loader_sync_cmd()`. This causes `detect_chip` to fail because it expects a response to its register read, but actually processes one of the duplicated packets.

`esptool` reads these 7 additional responses, see https://github.com/espressif/esptool/blob/master/esptool.py#L432 . Can we rely on always receiving 8 packets to a sync query? If so it's an easy fix, but not sure if different chips will have different behaviour. I'm a bit surprised that the code works for other devices at the moment, so let me know if I'm missing something :)

campbell-seer, 2020-07-21:
Read over the code a bit more and see that `check_response` handles this by ignoring commands that aren't expected. So the issue must be that my rx buffer is overflowing, and then the packets are no longer aligned and so this check doesn't work. Sorry for the bad report :+1:

closed 2020-07-21 by campbell-seer

MaValki, 2020-07-21:
Hi campbell-seer,
that is right, unwanted packets are ignored. `check_response` function only return in case of TIMEOUT or some other error.
If you have any additional questions, don't hesitate to ask.