espressif/esp-serial-flasher #11: Flash erase time is too short

campbell-seer, 2020-08-07:
I have an ESP32 that I program with three images in the following order:

- Partitions (about 3 KiB)
- Bootloader (about 24 KiB)
- App (about 700 KiB)

The bootloader and partition images write fine every time. However, the app fails when the flash is not currently erased and works if the flash is erased. 

According to `ERASE_REGION_TIMEOUT_PER_MB` the erase should take no more than 3 seconds. However, it actually takes about 5 seconds before the ESP32 responds that the erase is done. As such, `esp_loader_flash_start` times out and I cannot program the device. 

So it seems that `ERASE_REGION_TIMEOUT_PER_MB` is too low. In `esptool` a timeout of 30 seconds is used, which seems way too high. What should this value be? I've modified the code to set it to 10 seconds, but really have no idea.

MaValki, 2020-08-07:
Hi,
Erase time is dependent on attached flash chip. Please take look at https://www.esp32.com/viewtopic.php?t=10716.
Even though I have not encountered any issue regarding to erase time (with `ERASE_REGION_TIMEOUT_PER_MB` set to 3 seconds), you may have a point. One of the possible flash chips on your board may be GD25Q32. According to documentation, full chip (4 MB) is supposed to be erased in 15 seconds. With current `ERASE_REGION_TIMEOUT_PER_MB` value set to 3 seconds, timeout will occur after 12 seconds. I will change `ERASE_REGION_TIMEOUT_PER_MB` from 3 to 10 seconds.

Thank you for contribution.

jimparis, 2020-12-14:
On my ESP32-S2-WROOM module, a call to `esp_loader_flash_start` with a length of 949264 bytes takes 4.9 seconds to complete.  So bumping the timeout to 10 seconds per MB would work.

MaValki, 2020-12-14:
Hi Jim Paris,

thank you for contribution. I have just changed `ERASE_REGION_TIMEOUT_PER_MB` to 10 seconds. It will be merged to master soon.

closed 2020-12-14 by MaValki