espressif/esp-serial-flasher #12: Flash verification / MD5 for ESP8266/8285

Hoel, 2020-10-03:
Hi,
Is there any plan to support flash verification / MDU check for ESP8266/ESP8285 platform?
It is quite necessary to verify integrity after flashing, especially with UART and no flow control., in particular when used in production context (test jig).
Also is there any update to change current STM32 implementation from polling to DMA?

MaValki, 2020-10-05:
Hi Hoel,

First stage bootloader of ESP8266/ESP8285 does not support MD5 verification. `esptool` loads second stage bootloader (`stub`), adding MD5 check among other feature. Incorporating `stub` into `esp-serial-flasher` is on your to do list, however, original implementation from `esptool` cannot be used. Therefore, it most likely will not be supported in very near future. My rough estimation would be 2-3 months.

Please expect DMA implementation of STM32 port to be done in 3 weeks.

Hope this helps.

Hoel, 2020-10-14:
hello,
thank you for the prompt reply.
Oh yes i see, we will do without the MD5 for now.
As a side note, the problems encountered with STM32 non-dma UART implmentation were due to another MCU on the board, once this issue has been fixed the non-dma implmentation works well at 921600bps (it still seems a little slow) the only problem is that connection is not fully reliable, ocasionally it fails or take a little while to connect.

closed 2021-12-15 by MaValki