espressif/esp-serial-flasher #14: STM32 timeout on connect

JHedgesBartec, 2020-12-08:
Hello,

I am having trouble with getting the STM32 port to connect to the ESP32 on our custom hardware. The port is very much the same apart from being integrated into another application and different IO assignments.

If I put TeraTerm on the Tx of the ESP32 WROOM (GPIO1) I can see the following after loader_port_enter_bootloader():
_"÷þ€ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download"_

The UART returns a timeout from SLIP_send_delimiter() in loader_sync_cmd().

Here is our GPIO connection:
STM32F4 USART Tx - > ESP32 GPIO 3
STM32F4 USART Rx - > ESP32 GPIO 1
TARGET_RST - > ESP32 EN
TARGET_IO0 - > ESP32 GPIO0

The above IO is exposed to an FTDI connector of which can be used with the flash download tool to program the module successfully.

Is our GPIO connection correct to the ESP32, and if so do you have any pointers as to where to start.
I am very familiar with STM32 and HAL, not so much this library.

Many Thanks,
Jack

JHedgesBartec, 2020-12-08:
Apologies for the mis click in closing the issue.
Here is some debug trace that may help:

`--- WRITE ---
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
C0 
00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 
C0 
Cannot connect to target. Error: 2`

MaValki, 2020-12-09:
Hi JHedgesBartec,

Thank you for detailed description. Your ESP32 connection seem to be correct.
ESP32 entered bootloader after calling `loader_port_enter_bootloader()`, however, doesn't reply to synchronization command.

Do you have FTDI bridge connected to the board while trying to flash with `esp-serial-flasher` ?
If so, detach it and try again.

Regards,
Martin

JHedgesBartec, 2020-12-09:
Hi @MaValki,

I conntedted the FTDI bridge just to check I could see the ESP32 boot message "waiting for download", after that it was disconnected.
I have just tried on another peice of hardware and still get the same behaviour.

Putting the FTDI bridge on to read the Tx of the STM32 shows that the command is being sent:
[image]

Thanks,
Jack

JHedgesBartec, 2020-12-09:
Hi @MaValki,

It looks like having the SERIAL_DEBUG_ENABLE #define enabled in stm32_port.c causes the second command to timeout when transmitting over the UART. Disabling this now allows the target to be connected and the baudrate changed.

Thanks,
Jack.

MaValki, 2020-12-09:
Indeed SERIAL_DEBUG_ENABLE can cause significant delay depending on implementation. Assuming you use semihosting on STM32. Does this mean you can flash the target after successful connection?

JHedgesBartec, 2020-12-09:
We are using semihosting on the STM32.

Yes, it works. Programmed a couple of different application versions to verify and it is all OK.

Thankyou for the simple and easy to use library!

closed 2020-12-09 by JHedgesBartec