espressif/esp-serial-flasher #16: Feature request: Erase region

jeremyherbert, 2021-01-18:
It would be very useful to have a function to erase a region and/or all of the flash. My use case is erasing an NVS partition when flashing a new firmware to a device.

MaValki, 2021-01-18:
Hi Jeremy,

Bootloader in ROM does not implement this functionality, so in order to achieve this, fill (write) remaining space in partition with 0xFF. Esptool allows to erase region, by loading second stage bootloader (stub), that adds this functionality . We plan to add stub into esp-serial-flasher also, but not sure when exactly. I hope this was helpful. If you have any additional question, don't hesitate to ask.

Martin

jeremyherbert, 2021-01-18:
Thanks Martin. Would be good to see the stub added in the future.

closed 2021-01-18 by jeremyherbert