espressif/esp-serial-flasher #17: Can’t connect Esp-01 for flashing if grounded

Mrlonch, 2021-01-28:
Hi there!

I need some help to understand what am I doing wrong. I have an Esp-01 and a serial device to connect it through USB, iba e soldered two push buttons as in the picture below:
[image]

completed the second button and tested continuity and all seems to be ok, but when I run the button sequence to put the Esp to flash mode it seems arduino IDE cannot connect to the device. Serial monitor only works for 74800 baud rate and all I can read there is a fatal error message when connecting.

additionally the Esp has two led lights, one red and one blue, if I press the reset the blue light flashes and stays, if I press reset and then flash, the blue light turns off and the red stays. But none of the combinations I have tried with the buttons seems to work...

Any clue??

Many thanks in advance!

MaValki, 2021-01-28:
Hi,
Unfortunately, I don't have ESP-01 at my disposal, so I cannot debug it right now.
Were you able to flash ESP-01 before with USB <-> UART bridge? If so, what pins did you connect ?
In theory, just RX, TX, IO0, RESET should be connected for putting ESP into boot mode and flash it. On ESP-01 board, however, GPIO2  and CH_PU should be pulled low or high.

Mrlonch, 2021-01-28:
Thanks for the time!

I have not been able to load any piece of code yet to the Esp-01, cannot set it to flash mode.

I soldered GND to GPIO0 and bridge it with the second button to reset, here the completed buttons:
[image]

if I connect the serial pressing the first button, the blue led goes away and the red stays:
[image]

if I click reset button, then blue led comes back again:
[image]

so not sure ....

MaValki, 2021-01-28:
According to what I read,  RED Led is connected to Power, while the BLUE Led is connected to TX. That doesn't tell us whether ESP is in boot mode or not. Seemingly, this programmer is notorious for not working.
ESP8266 enters bootloader after reset when IO0 is held low. Therefore, Press RESET button while holding IO0 button down.
Although, naming conventions here are a bit misleading, I think that CH_PU pin is resetting the chip. So you might need to connect button to that pin instead, or follow this article (in that case, you don't need reset button at all): 
https://makeradvisor.com/esp8266-esp-01-usb-serial-programmer/

I hope that hepls

Mrlonch, 2021-01-28:
It helped me to understand a bit better, but still not switching to flash mode... any other check I could do?? My button should work as in the post provided but not sure if it is a bad connection of the wires, but checked with multimeter and button does it’s function... completely lost now...

MaValki, 2021-01-28:
Do you see anything in terminal (some serial monitor) after pressing RESET (CH_PU)? Assuming you can at least connect to serial port.

Mrlonch, 2021-01-28:
Yes, but only at 74800 baud rate and the message I see is something like “fatal error”. So I also think I cannot connect properly to the serial...

Mrlonch, 2021-01-28:
I will try connecting the Esp-01 through a lon arduino mega and see if there I manage to get it working...maybe that points to the serial device malfunctioning...

closed 2021-03-03 by MaValki