espressif/esp-serial-flasher #18: How to set SpiFlashConfig?

hehao9051, 2021-01-31:
When i use esp_download tool, i need set spiflashconfig. But in this project, i can't find how to set it. 
[image]

MaValki, 2021-02-01:
Hi,

FLASH  SIZE of attached flash chip is detected automatically, however, modifying SPI SPEED and SPI MODE is unimplemented. Those parameters are included at the beginning of generated binary and can be selected in menuconfig (_serial flash config_ submenu). Do you have to override it while flashing for some reason ?

Hope this helps.

hehao9051, 2021-02-01:
@MaValki  Thanks we don't need that. Because we alaway use the tools, so i mistakenly think the config have to set it by flasher.

closed 2021-02-01 by hehao9051