espressif/esp-serial-flasher #21: ESP_LOADER_ERROR_INVALID_RESPONSE with STM32 port

endlos99, 2021-03-14:
I'm trying to use esp-serial-flasher with an STM32F7 MCU to flash an ESP8266.  Unfortunately, I always get `ESP_LOADER_ERROR_INVALID_RESPONSE` right in the connection phase.

The biggest issue IMHO is that you're receiving each character individually, which probably causes characters to drop while `HAL_UART_Receive` is not active.  I've remedied this in my code by issuing one `HAL_UART_Receive_DMA` before calling esp-serial-flasher, and then in `loader_port_serial_read`, I check the receiver buffer for new characters.  That way, nothing sent by the ESP will be lost.  Unfortunately, I'm still getting `ESP_LOADER_ERROR_INVALID_RESPONSE`.

In function `SLIP_receive_packet`, you wait for _two_ characters 0xc0, although there is a comment "Workaround: bootloader sends two dummy(0xc0) bytes after response when baud rate is changed." for the second char.  The ESP8266 doesn't support baud rate changes, which is why `loader_change_baudrate_cmd` is not called.  But doesn't that conflict `SLIP_receive_packet`?

For what it's worth, here are the first 40 bytes that I receive from the ESP (after the `loader_sync_cmd`).  Note that there is only one 0xc0 char at the beginning, not two.

[image]

Still, there are too many chars between the delimiters, so esp-serial-flasher still throws `ESP_LOADER_ERROR_INVALID_RESPONSE`.

Could you please comment on the byte sequence received and the issues raised here?

YWHyuk, 2021-10-19:
Hi @endlos99
Although the issue has been registered for a long time, I have a similar issue. 
I'm using NUCLEO-H723ZG board and used the STM32 example of git repository.

In my case, I returned `ESP_LOADER_ERROR_TIMEOUT`.

When I activate UART fifo feature, it works. 
This is my UART init function.
```C
static void MX_USART3_UART_Init(void)
{
  huart3.Instance = USART3;
  huart3.Init.BaudRate = 115200;
  huart3.Init.WordLength = UART_WORDLENGTH_8B;
  huart3.Init.StopBits = UART_STOPBITS_1;
  huart3.Init.Parity = UART_PARITY_NONE;
  huart3.Init.Mode = UART_MODE_TX_RX;
  huart3.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart3.Init.OverSampling = UART_OVERSAMPLING_16;
  huart3.Init.OneBitSampling = UART_ONE_BIT_SAMPLE_DISABLE;
  huart3.Init.ClockPrescaler = UART_PRESCALER_DIV1;
  huart3.AdvancedInit.AdvFeatureInit = UART_ADVFEATURE_NO_INIT;
  if (HAL_UART_Init(&huart3) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_SetTxFifoThreshold(&huart3, UART_TXFIFO_THRESHOLD_1_2) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_SetRxFifoThreshold(&huart3, UART_RXFIFO_THRESHOLD_1_2) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_EnableFifoMode(&huart3) != HAL_OK)
  {
    Error_Handler();
  }
}
```

danergo, 2022-01-16:
Hi! I'm facing with the same problem on RPi3, when migrating from RPi1.

On RPi1 flashing works perfect, but on RPi3 it always fails.

Does anyone have an idea on how can I solve this on faster RPis?

Thank you!

danergo, 2022-02-07:
ping

YWHyuk, 2022-02-07:
Hi @danergo 
Can you describe your situation with more detailed? As you know, this issue was opened from STM32 implementation.

danergo, 2022-02-07:
Hi @YWHyuk: sure, thank you for the reaction!

Shall I summarize it here, or open a new ticket?

danergo, 2022-02-07:
Sorry, it was my defected SPI kernel driver. Works perfectly with RPi's onboard stuff.

DNedic, 2023-07-17:
Hello @endlos99 , sorry for the late response, we just recently started tracking `esp-serial-flasher` issues. A fix for #65 was just merged, and there is a possibility that this also fixes your issue. Please retest if you are still having the issue with the latest fixes.

DNedic, 2023-08-05:
I'm closing this for inactivity.

closed 2023-08-05 by DNedic