espressif/esp-serial-flasher #23: ESP32 MD5 calculation command loader_md5_cmd() does not work for above 2 Megabytes region (0x00200000)

stsymbaliuk, 2021-04-13:
I use STM32 port. My binary file size is 2428544 bytes (2.31 MB). I can flash it, but I have problem with MD5 verify.
I found that `loader_md5_cmd()` can check only first 2 MB of flash.

For address above 0x00200000
- ESP32 returns packet with `FLASH_READ_ERR`;
- `loader_md5_cmd()` returns `ESP_LOADER_ERROR_INVALID_RESPONSE` (9);

I can read back 2428544 bytes of my firmware binary file from ESP32 with `esptool` and it is OK
```
esptool.py --chip esp32 --port /dev/ttyS7 --baud 921600 read_flash 0 2428544 read.bin
```

Also I can use `loader_md5_cmd()` for address `0` and size `0x00200000` and compare it to MD5 calculated on PC - also OK.

Looks like ESP32 cannot calculate MD5 for that memory region.

My test function for `loader_md5_cmd()` is based on `esp_loader_flash_verify()`:
```c
esp_loader_error_t esp_loader_flash_md_calc(size_t address, size_t size)
{
    printf("%s() address: 0x%08zx, size: %7zu, ", __func__, address, size);

    if (s_target == ESP8266_CHIP) {
        return ESP_LOADER_ERROR_UNSUPPORTED_FUNC;
    }

    uint8_t received_md5[MD5_SIZE + 2] = {0};

    loader_port_start_timer(timeout_per_mb(size, MD5_TIMEOUT_PER_MB));

    esp_loader_error_t err = loader_md5_cmd(address, size, received_md5);
    if (err != ESP_LOADER_SUCCESS)
    {
    	printf("Error: %d\n", err);
      return err;
    }

    received_md5[MD5_SIZE] = '\n';

    printf("Received MD5: ");
    printf((char *)received_md5);

    return ESP_LOADER_SUCCESS;
}
```

My ESP32 has 4MB flash chip.

flash_download_tool_3.8.5 output:
```
flash vendor:
C8h : GD
flash devID:
6016h
N/A;32Mbit
crystal:
40 Mhz

AP:  CC50E3AAC43D  STA:  CC50E3AAC43C
BT:  CC50E3AAC43E  ETHERNET:  CC50E3AAC43F
```

```
esptool.py --chip esp32 --port /dev/ttyS7 --baud 921600 flash_id
esptool.py v2.8
Serial port /dev/ttyS7
Connecting........_____.....__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: cc:50:e3:aa:c4:3c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Manufacturer: c8
Device: 6016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

Next are logic analyzer captured requests and responses.

MD5 request and response for 0x001FF800 address, 0x800 size - Ok
```
Time [s],Value,Parity Error,Framing Error
0.000000000,0xC0,,
0.000024000,0x00,, // WRITE_DIRECTION
0.000034900,0x13,, // SPI_FLASH_MD5
0.000045900,0x10,, // size
0.000056800,0x00,, // size
0.000067700,0x00,, // checksum
0.000078600,0x00,, // checksum
0.000089600,0x00,, // checksum
0.000100500,0x00,, // checksum
0.000111400,0x00,, // address
0.000122300,0xF8,, // address
0.000133300,0x1F,, // address
0.000144200,0x00,, // address
0.000155100,0x00,, // size
0.000166000,0x08,, // size
0.000177000,0x00,, // size
0.000187900,0x00,, // size
0.000198800,0x00,, // reserved_0
0.000209800,0x00,, // reserved_0
0.000220700,0x00,, // reserved_0
0.000231600,0x00,, // reserved_0
0.000242500,0x00,, // reserved_1
0.000253500,0x00,, // reserved_1
0.000264400,0x00,, // reserved_1
0.000275300,0x00,, // reserved_1
0.000291500,0xC0,,

Time [s],Value,Parity Error,Framing Error
0.003226900,0xC0,,
0.003237500,0x01,, // READ_DIRECTION
0.003248200,0x13,, // SPI_FLASH_MD5 command
0.003258900,0x24,, // size 
0.003269500,0x00,, // size 
0.003280200,0x00,, // value
0.003290900,0xA0,, // value
0.003301600,0x00,, // value
0.003312200,0x00,, // value
0.003322900,0x38,, // md5 [0]
0.003333600,0x62,,
0.003344300,0x31,,
0.003354900,0x32,,
0.003365600,0x61,,
0.003376300,0x30,,
0.003386900,0x64,,
0.003397600,0x66,,
0.003408300,0x66,,
0.003418900,0x38,,
0.003429600,0x30,,
0.003440300,0x30,,
0.003451000,0x61,,
0.003461600,0x61,,
0.003472300,0x30,,
0.003483000,0x38,,
0.003493700,0x63,,
0.003504300,0x34,,
0.003515000,0x66,,
0.003525700,0x30,,
0.003536300,0x65,,
0.003547000,0x35,,
0.003557700,0x36,,
0.003568400,0x37,,
0.003579000,0x30,,
0.003589700,0x31,,
0.003600400,0x66,,
0.003611000,0x30,,
0.003621700,0x34,,
0.003632400,0x64,,
0.003643100,0x37,,
0.003653700,0x62,, // md5 [31]
0.003664400,0x00,, // failed
0.003675100,0x00,, // error
0.003685800,0x00,,
0.003696400,0x00,,
0.003707100,0xC0,,
```

MD5 request and response for 0x001FFC00 address, 0x800 size - Fail
```
Time [s],Value,Parity Error,Framing Error
0.000000000,0xC0,,
0.000024000,0x00,, // WRITE_DIRECTION
0.000034900,0x13,, // SPI_FLASH_MD5
0.000045800,0x10,, // size
0.000056800,0x00,, // size
0.000067700,0x00,, // checksum
0.000078600,0x00,, // checksum
0.000089500,0x00,, // checksum
0.000100500,0x00,, // checksum
0.000111400,0x00,, // address
0.000122300,0xFC,, // address
0.000133200,0x1F,, // address
0.000144200,0x00,, // address
0.000155100,0x00,, // size
0.000166000,0x08,, // size
0.000176900,0x00,, // size
0.000187900,0x00,, // size
0.000198800,0x00,, // reserved_0
0.000209700,0x00,, // reserved_0
0.000220600,0x00,, // reserved_0
0.000231600,0x00,, // reserved_0
0.000242500,0x00,, // reserved_1
0.000253400,0x00,, // reserved_1
0.000264400,0x00,, // reserved_1
0.000275300,0x00,, // reserved_1
0.000291500,0xC0,,

Time [s],Value,Parity Error,Framing Error
0.001755100,0xC0,,
0.001765700,0x01,, // READ_DIRECTION
0.001776400,0x13,, // SPI_FLASH_MD5 command
0.001787100,0x24,, // size 
0.001797700,0x00,, // size 
0.001808400,0x00,, // value
0.001819100,0xA0,, // value
0.001829800,0x00,, // value
0.001840400,0x00,, // value
0.001851100,0x38,, // md5 [0]
0.001861800,0x62,,
0.001872500,0x31,,
0.001883100,0x32,,
0.001893800,0x61,,
0.001904500,0x30,,
0.001915100,0x64,,
0.001925800,0x66,,
0.001936500,0x66,,
0.001947100,0x38,,
0.001957800,0x30,,
0.001968500,0x30,,
0.001979200,0x61,,
0.001989800,0x61,,
0.002000500,0x30,,
0.002011200,0x38,,
0.002021900,0x63,,
0.002032500,0x34,,
0.002043200,0x66,,
0.002053900,0x30,,
0.002064500,0x65,,
0.002075200,0x35,,
0.002085900,0x36,,
0.002096600,0x37,,
0.002107200,0x30,,
0.002117900,0x31,,
0.002128600,0x66,,
0.002139200,0x30,,
0.002149900,0x34,,
0.002160600,0x64,,
0.002171200,0x37,,
0.002181900,0x62,, // md5 [31]
0.002192600,0x01,, // failed
0.002203200,0x09,, // error
0.002213900,0x00,,
0.002224600,0x00,,
0.002235300,0xC0,,
```

MD5 request and response for 0x00200000 address, 0x400 size - Fail
```
Time [s],Value,Parity Error,Framing Error
0.000000000,0xC0,,
0.000023600,0x00,, // WRITE_DIRECTION
0.000034600,0x13,, // SPI_FLASH_MD5
0.000045500,0x10,, // size
0.000056400,0x00,, // size
0.000067300,0x00,, // checksum
0.000078300,0x00,, // checksum
0.000089200,0x00,, // checksum
0.000100100,0x00,, // checksum
0.000111000,0x00,, // address
0.000122000,0x00,, // address
0.000132900,0x20,, // address
0.000143800,0x00,, // address
0.000154700,0x00,, // size
0.000165700,0x04,, // size
0.000176600,0x00,, // size
0.000187500,0x00,, // size
0.000198400,0x00,, // reserved_0
0.000209400,0x00,, // reserved_0
0.000220300,0x00,, // reserved_0
0.000231200,0x00,, // reserved_0
0.000242200,0x00,, // reserved_1
0.000253100,0x00,, // reserved_1
0.000264000,0x00,, // reserved_1
0.000274900,0x00,, // reserved_1
0.000290500,0xC0,,

Time [s],Value,Parity Error,Framing Error
0.000319600,0xC0,,
0.000330300,0x01,,
0.000341000,0x13,,
0.000351700,0x24,,
0.000362300,0x00,,
0.000373000,0x00,,
0.000383700,0xA0,,
0.000394300,0x00,,
0.000405000,0x00,,
0.000415700,0x38,,
0.000426300,0x62,,
0.000437000,0x31,,
0.000447700,0x32,,
0.000458400,0x61,,
0.000469000,0x30,,
0.000479700,0x64,,
0.000490400,0x66,,
0.000501100,0x66,,
0.000511700,0x38,,
0.000522400,0x30,,
0.000533100,0x30,,
0.000543700,0x61,,
0.000554400,0x61,,
0.000565100,0x30,,
0.000575800,0x38,,
0.000586400,0x63,,
0.000597100,0x34,,
0.000607800,0x66,,
0.000618400,0x30,,
0.000629100,0x65,,
0.000639800,0x35,,
0.000650500,0x36,,
0.000661100,0x37,,
0.000671800,0x30,,
0.000682500,0x31,,
0.000693200,0x66,,
0.000703800,0x30,,
0.000714500,0x34,,
0.000725200,0x64,,
0.000735800,0x37,,
0.000746500,0x62,,
0.000757200,0x01,, // failed
0.000767800,0x09,, // error
0.000778500,0x00,,
0.000789200,0x00,,
0.000799900,0xC0,,
```

huybk213, 2022-06-27:
I have the same issues

LnnrtS, 2023-12-01:
If I write to a lower region (`0x00010000 - 0x00169950` for example) before checking the md5 for `0x00200000` and up it works. Just getting the checksum of a lower region is not enough though

DNedic, 2023-12-01:
Hello @LnnrtS , thank you for providing more insight on the problem, I will investigate further.

DNedic, 2023-12-19:
I could not reproduce the issue. I have managed to flash apps over 3MB in size to both the ESP32 and the ESP32-C3 and verify the flash using MD5 without a problem.

> Can you read the same region with esptool and without the stub? For me this fails while it works with the stub

@LnnrtS On the first glance, It seems to me that this is an unrelated issue, I will investigate it. 

I've fixed the MD5 timeout issue I was facing https://github.com/espressif/esp-serial-flasher/commit/058156c083c4e059ee7d0ec99248b53cfb41be41, this did not fail in the same way as for the original poster, but perhaps you can retry and see if it fixed anything for you?

LnnrtS, 2023-12-20:
> I could not reproduce the issue. I have managed to flash apps over 3MB in size to both the ESP32 and the ESP32-C3 and verify the flash using MD5 without a problem.

The problem appears only when all of these are met:
- only verify is run, no flashing before
- stub is not used
- targeting a region above `0x00200000`

Did you try exactly that? In particular, when flash before in the same session the error will not appear

> On the first glance, It seems to me that this is an unrelated issue, I will investigate it.

Could be. However, the conditions for a failure and the failure itself are very similar. 

> I've fixed the MD5 timeout issue I was facing https://github.com/espressif/esp-serial-flasher/commit/058156c083c4e059ee7d0ec99248b53cfb41be41, this did not fail in the same way as for the original poster, but perhaps you can retry and see if it fixed anything for you?

I already had this fixed in my fork before. It's a differrent issue with a different failure (timeout).

DNedic, 2024-01-25:
Hello @LnnrtS , currently verifying without flashing is not a part of the public API, although we might add it in the future. We will test out the failure you described before going ahead with it if we decide to.

Since this issue is related to the public API flashing and I could not reproduce the problems, I will close it for now and we can reopen it if someone else finds a similar problem.

closed 2024-01-25 by DNedic

jcwo1975, 2024-02-05:
Hi all, calling the below functions before using the loader_md5_cmd() seems able to solve the issue on my side. See if you can get the same positive result from it.

```
    size_t flash_size = 0;
    if (detect_flash_size(&flash_size) == ESP_LOADER_SUCCESS)
    {
        loader_port_start_timer(DEFAULT_TIMEOUT);
        RETURN_ON_ERROR( loader_spi_parameters(flash_size) );
    }
```

LnnrtS, 2024-02-06:
Thanks, the fix works for me!

DNedic, 2024-02-06:
Interesting, I will investigate why this is happening and we will take this into consideration if we decide to add the preexisting md5 verification to the public API. 

> Looking at the code I would expect this should also happen when calling esp_loader_flash_verify which is part of the public API.

True however it is not designed so that you can call it before something is already flashed and will not work.