espressif/esp-serial-flasher #28: Arduino Nano 33 IoT port

beriberikix, 2021-07-11:
https://store.arduino.cc/usa/nano-33-iot

The Arduino Nano 33 IoT is a low-cost Arduino board with a SAMD21 host processor and U-Blox NINA modem. The firmware is [custom](https://github.com/arduino/nina-fw) with an [Arduino tool](https://www.arduino.cc/en/Tutorial/WiFiNINAFirmwareUpdater) to update it.

Are there plans to target Microchip hosts like the SAMD21? If not, are there any examples in the community to look at?

MaValki, 2021-07-12:
Hi beriberikix,

I didn't have intentions to add support for MIcrochip MCUs, as no one has asked for it before.
However, creating Arduino library on top of some platform independent UART and GPIO libraries seems to be a good idea.

I don't know about any examples with Microchip MCUs.
At the moment, I am also working on other projects, so I am not able to give you any time line when Arduino support will be ready.

In case you need port in near future, adding support for new target should be relatively easy. See README for reference what functions has to be implemented.

Martin

winner10920, 2021-09-14:
Hi, I've managed to mangle together a port* for the Arduino nano 33 iot. If your interested I'll make it a bit more readable and share the code, it would require quite a bit of work to fit the thematics of this git, 

*it's not really a port but I took this source and turned it into an Arduino library for use with the Arduino IDE. I'm assuming all this is done with cmake (which I don't know how to use really) and the only functional testing I've done is within the Arduino IDE

MaValki, 2021-09-15:
Hi  winner10920, 

if you can put together some working version of this port, your contribution in form of pull request would be much appreciated.

beriberikix, 2023-02-14:
I haven't tested this yet but #34 should be able to address this issue!