espressif/esp-serial-flasher #30: Build bootloader.bin

SantiFerre, 2021-10-05:
There is any option to build “bootloader.bin” from platformio using ESP32 and Arduino framework?

I will flash one ESP32 from another ESP32 based on the examples, it works with hello-world because have a correct bootloader.
If I flash my firmware directly from platformio and then I flash firmware and partitions from the other ESP32, it works fine. But when I flash it with the bootloader.bin (bootloader_dio_40m.bin). It comes a constant loop:

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ets Jul 29 2019 12:21:46
```

Thanks in advance.

SantiFerre, 2021-10-11:
Hi,
I'm using the example with my own modifications:
```
FILE: example_common.c

#define BOOTLOADER_ADDRESS  0x1000
#define PARTITION_ADDRESS   0x8000
#define APPLICATION_ADDRESS 0x10000

extern const uint8_t firmware_bin_start[] asm("_binary_src_firmware_firmware_bin_start");
extern const uint8_t firmware_bin_end[] asm("_binary_src_firmware_firmware_bin_end");

extern const uint8_t bootloader_bin_start[] asm("_binary_src_firmware_bootloader_bin_start");
extern const uint8_t bootloader_bin_end[] asm("_binary_src_firmware_bootloader_bin_end");

extern const uint8_t partitions_bin_start[] asm("_binary_src_firmware_partitions_bin_start");
extern const uint8_t partitions_bin_end[] asm("_binary_src_firmware_partitions_bin_end");

void get_example_binaries(target_chip_t target, example_binaries_t *bins) {
    
    bins->app.data = firmware_bin_start;
    bins->app.size = (firmware_bin_end - firmware_bin_start) - 1;
    bins->app.addr = APPLICATION_ADDRESS;

    bins->boot.data = bootloader_bin_start;
    bins->boot.size = (bootloader_bin_end - bootloader_bin_start) - 1;
    bins->boot.addr = BOOTLOADER_ADDRESS;

    bins->part.data = partitions_bin_start;
    bins->part.size = (partitions_bin_end - partitions_bin_start) - 1;
    bins->part.addr = PARTITION_ADDRESS;
}
```

```
FILE: main.cpp

if (connect_to_target(HIGHER_BAUDRATE) == ESP_LOADER_SUCCESS) {
    get_example_binaries(esp_loader_get_target(), &bin);

    flash_binary(bin.boot.data, bin.boot.size, bin.boot.addr);
    flash_binary(bin.part.data, bin.part.size, bin.part.addr);
    flash_binary(bin.app.data,  bin.app.size,  bin.app.addr);
}
```

What file should I load into the bootloader?

GodayMax, 2022-01-11:
Same issue, placed partions.bin and firmware.bin not worked

E (982) esp_image: Image length 2308192 doesn't fit in partition length 1048576
E (987) boot: Factory app partition is not bootable
E (992) boot: No bootable app partitions in the partition table

dobairoland, 2023-05-26:
I'm closing this for inactivity and because it wasn't demonstrated any clear issue with esp-serial-flasher.

closed 2023-05-26 by dobairoland