espressif/esp-serial-flasher #31: ESP32C3 chip magic number missing

cmsedore, 2021-12-06:
Flagging this for others so they don't have to spend an hour chasing this down:

The flasher doesn't include both magic numbers for the ESP32C3: 0x1b31506f is what my C3s yield vs 0x6921506f in the file.  This block is the challenging one.  It appears you can't simply duplicate with the correct ID it as it is tied to an enum that appears in a couple of other places.

```
    // ESP32C3
    {
        .regs = {
            .cmd  = ESP32C3_SPI_REG_BASE + 0x00,
            .usr  = ESP32C3_SPI_REG_BASE + 0x18,
            .usr1 = ESP32C3_SPI_REG_BASE + 0x1c,
            .usr2 = ESP32C3_SPI_REG_BASE + 0x20,
            .w0   = ESP32C3_SPI_REG_BASE + 0x58,
            .mosi_dlen = ESP32C3_SPI_REG_BASE + 0x24,
            .miso_dlen = ESP32C3_SPI_REG_BASE + 0x28,
        },
        .efuse_base = 0x60008800,
        .chip_magic_value = 0x6921506f,     ////////////// mine required 0x1b31506f
        .read_spi_config = spi_config_esp32xx,
    },
```

https://github.com/espressif/esp-serial-flasher/blob/e44d50e3726c2faea93071762ce046ac4f91a75e/src/esp_targets.c#L102

MaValki, 2021-12-06:
Hi,
thank you for pointing that out. I have not noticed, that ESP32-C3 revision 3 has different magic number compared to rev. 1 and 2.
I'll add support for both. I am not quite sure what you meant by correct ID tied to an enum. Could you please clarify ?

cmsedore, 2021-12-06:
Sure--this is a member of this array:

```
static const esp_target_t esp_target[ESP_MAX_CHIP] = {
[...]
```

which I think is tied to this (from esp_loader.h):

closed 2021-12-15 by espressif-bot