espressif/esp-serial-flasher #35: Erase All Command

krdarrah, 2022-01-12:
Hello, I'm working on using this in a standalone factory programmer, but would like it to fully erase the ESP32 - how to implement the "ESP_ERASE_FLASH" command? My poor attempt returns "COMMAND FAILED" when I try sending 0xD0: 
```
esp_loader_error_t loader_eraseAllFlash_cmd()
{
    eraseAllCommad_t eraseAllCommad = {
        .common = {
            .direction = WRITE_DIRECTION,
            .command = 0xD0,
            .size = CMD_SIZE(eraseAllCommad),
            .checksum = 0
        },
    };

    return send_cmd(&eraseAllCommad, sizeof(eraseAllCommad), NULL);
}
```

igrr, 2022-01-14:
(linking the similar issue, https://github.com/espressif/esp-serial-flasher/issues/16)

Hi @krdarrah, the bootloader in ESP32 mask ROM doesn't support this command. In esptool.py it works by preloading the [flasher stub](https://github.com/espressif/esptool/tree/master/flasher_stub) into the IRAM and talking to the stub instead of the ROM bootloader.
`esp-serial-flasher` currently doesn't support loading the flasher stub into the target IRAM.

krdarrah, 2022-01-15:
thanks!  hopefully this will be supported at some point

closed 2022-01-15 by krdarrah