espressif/esp-serial-flasher #44: Question | Is there are something similar for setting/getting efuses?

Aukstkalnis, 2022-09-09:
I am looking for library to set ant read efuses. Is there something like this project to control efuses?

MaValki, 2022-09-12:
Hi, I am afraid there is no such project. Probably the simplest way to achieve this, would be to follow  `eFuse Controller` section in `TRM`(Technical Reference Manual) and manually write/read registers using `esp_loader_write_register`, `esp_loader_read_register` functions.

Aukstkalnis, 2022-09-13:
esp_loader_write_register and esp_loader_read_register is for internal reading of eFuses so it is not a solution for me. I am developing testing environment where FW is flashed over other MCU and I want to manually set eFuses to speed encryption process. But thanks for advice :)

hyzgit, 2023-02-22:
The function of configuring efuses is really important, such as when flash encryption is enabled.

Aukstkalnis, 2023-02-22:
> The function of configuring efuses is really important, such as when flash encryption is enabled.

What do you mean by that? The question is if there is a library that can set eFuses from, for example, stm32 MCU.

hyzgit, 2023-02-22:
> 

stm32 encrypted flash does not need to be configured with a eFuses  (and there is no eFuses )，However, to enable flash encryption on esp32, eFuses must be configured for it to work. If you download the program through esp-serial-flasher, then you have to take it to your computer to configure eFuses

https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/security/flash-encryption.html

Aukstkalnis, 2023-02-22:
The problem is that self flash encryption takes to much time to complete in mass production. So I am exploring an idea to encrypt firmware partitions, that has to be encrypted, then flash that image, then set encryption bits and security bits and in the end run ESP32 FW. This could save plenty of time in mass production. The device that controls flashing is STM32F407. Device with ESP32 is connected to STM32 and STM32 is connected to local network with ethernet.

sergiosider, 2023-10-27:
Write fuses directly would be a blessing. I can leave a simple esp32 wroom board with production without the need for a PC. A kind of temporary solution is to flash a "pre-program" into the target device and make him write the adequate fuses and the "writer" would then write the correct program.

sergiosider, 2023-10-27:
bullshit... @MaValki  is right. it's actualy easy:
I am using ESP32-S3 to write to a ESP32, so I had to include:

`
#include "../../esp32/include/soc/efuse_reg.h"
`

A simple write function would be like this:

`
esp_err_t My_program_fuse(uint32_t address, uint32_t value)
{
    esp_err_t ret = 1;
    uint32_t val;
    if(esp_loader_write_register(EFUSE_CLK_REG,0x00008050)) return 1; //EFUSE_CLK
    if(esp_loader_write_register(EFUSE_DAC_CONF_REG,100)) return 1; //EFUSE_DAC_CONF

    if(esp_loader_write_register(address,value)) return 1; 
    if(esp_loader_write_register(EFUSE_CONF_REG,0x5A5A)) return 1; //EFUSE_CONF
    if(esp_loader_write_register(EFUSE_CMD_REG,2)) return 1; //EFUSE_CMD
    do {
        if(esp_loader_read_register(EFUSE_CMD_REG,&val)) return 1;
        vTaskDelay(1);
    } while (val);

    if(esp_loader_write_register(EFUSE_CONF_REG,0x5AA5)) return 1; //EFUSE_CONF
    if(esp_loader_write_register(EFUSE_CMD_REG,1)) return 1; //EFUSE_CMD
    do {
        if(esp_loader_read_register(EFUSE_CMD_REG,&val)) return 1;
        vTaskDelay(1);
    } while (val);

    if(esp_loader_write_register(address,0)) return 1; 

    return 0; //ok

}
`