espressif/esp-serial-flasher #46: Timeout on loader_detect_chip

alepez, 2022-11-11:
My host is an STM32H743

My target is a ESP32-WROOM-32U

I can flash it using esptool, this is its output:

```txt
esptool.py v4.3
Serial port /dev/ttyUSB0
Connecting...
Chip is ESP32-D0WD (revision v1.0)
```

When trying to flash using the STM32H743, I got a timeout error. The connection seems ok, because the debugger shows that it reaches `RETURN_ON_ERROR( loader_detect_chip(&s_target, &s_reg) );`.

```c
    loader_port_enter_bootloader();

    do {
        loader_port_start_timer(connect_args->sync_timeout);
        err = loader_sync_cmd();
        if (err == ESP_LOADER_ERROR_TIMEOUT) {
            if (--trials == 0) {
                return ESP_LOADER_ERROR_TIMEOUT;
            }
            loader_port_delay_ms(100);
        } else if (err != ESP_LOADER_SUCCESS) {
            return err;
        }
    } while (err != ESP_LOADER_SUCCESS);

    RETURN_ON_ERROR( loader_detect_chip(&s_target, &s_reg) );
```

I get an `ESP_LOADER_ERROR_TIMEOUT` error inside `check_response` called by `loader_detect_chip`.

- I'm sure the target is in boot mode (if I do not put it in boot mode, I get a timeout error early)
- I'm using STM32 FIFO mode

This is the record of received data:

```
C0 01 08 04 00 07 12 20 55 00 00 00 00 C0 C0 01
08 04 00 07 12 20 55 00 00 00 00 C0 C0 01 08 C0
01 0A 04 00
```

Any idea?

DNedic, 2023-05-28:
Hello @alepez, is this all the data received? Do you have by any chance a logic analyzer trace? 

The data received here is just sync responses but the last one is cut off, not enough to infer where the fault lies.

alepez, 2023-06-26:
Hello @DNedic, I'm sorry but many months have passed since then and I don't remember what was causing this issue.

Anyway, it's currently working with version baa1a03 without any changes, so it was some bug on my side.

DNedic, 2023-06-26:
Glad to hear that @alepez , please report any more issues you might have as we are tracking them in our internal system now and we can respond quickly.