espressif/esp-serial-flasher #53: Adding flasher stub to STM32 serial flasher

JeffreyEWright, 2022-12-11:
Hello;
I was successful porting the serial flasher for my client's stm32l432, including MD5 support. I then included code to upload the stub flasher, but I cannot seem to get it to run...the esp32c3 always crashes after I send the MEM_END command. 
It seems that either I'm not writing to the correct addresses in IRAM, OR I'm reading the .bin file at the wrong address(es)...I can read back what I've written (MEM_DATA, READ_REG)

If I understand correctly as explained here ->
https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/firmware-image-format.html#image-format
the 8 byte header is followed by a 16 byte header, and then the 2, 8-byte segment information blocks. 

As per the headers in the .bin file (see in image), I write the 'text' segment starting at offset 0x0x2C in the .bin file, starting at address 0x40380000 in the ESP32C3's IRAM. I then write the 4 byte 'data' segment to address 0x3FC96BA8. This returns no errors. Finally, I issue the MEM_END command, using the entry point address, 0x40380670, at which point the esp32 crashes. 

[image]
Question: am I using the correct addresses in the IRAM and the .bin file?
Any ideas would be greatly appreciated!

dobairoland, 2022-12-12:
Hi @JeffreyEWright.

I think it could help you a lot if you would try the same with [esptool.py](https://github.com/espressif/esptool/) and examine the output when using the `--trace` argument.

The IRAM address is the same as the one in esptool's flasher stub: https://github.com/espressif/esptool/blob/master/flasher_stub/ld/stub_32c3.ld#L8. DRAM and the entry point (https://github.com/espressif/esptool/blob/master/esptool/targets/stub_flasher/stub_flasher_32c3.json#L2) is a little different but withing reasonable limits.

JeffreyEWright, 2022-12-13:
Just as I got the stub flasher to load and run successfully, now there's another problem...once the stub is running, it configures the SPI ok, but then returns an unknown error after I send it the loader_flash_begin_cmd()??
The ROM bootloader **works OK**, but not when running the stub.
The documentation says the error codes for the stub all start with '0xC* or 0xFF', ESP_OK = 0, but you can see in the attached capture that the status bytes are 0x01, 0x0 ?????
Any ideas?
THanks!
[image]

JeffreyEWright, 2022-12-13:
I found the troubles: 
1. begin_command_t structure is incompatible with the stub loader.  I created a new one omitting the  uint32_t encrypted member;
2. MD5_SIZE must be changed to 16
4. After all the pain, it only drops 10 seconds off the flash time...disappointing

closed 2022-12-13 by JeffreyEWright