espressif/esp-serial-flasher #54: Document the maximum allowed block size

not-na, 2023-01-17:
I could not find anywhere the maximum allowed block size for `esp_loader_flash_start()`  is documented. From my testing,
4096 bytes works fine, but larger sizes give an `INVALID_COMMAND` error without further explanation

It would be nice to document the actual limit somewhere it can be easily found. This would've prevented a lot of debugging for me, since I assumed that I could just use a larger buffer without issues.

If you want, I can make a PR that adds a corresponding note to `esp_loader.h` [here](https://github.com/espressif/esp-serial-flasher/blob/a61f7449d7d3c4e63e5188c398b6ea1b1d35a002/include/esp_loader.h#L127).

dobairoland, 2023-01-18:
Hi @not-na. I went though the sources and don't think that is a limitation set on the esp-serial-flasher side. The SLIP packet gets successfully sent and the INVALID_COMMAND originates from the response. In our examples 1024 is used which seems to be a safe value. I would not put anywhere the limit you've experimentally got because IMO it is entirely deduced from the flash chip. I cannot tell if it would be the same for other chips.

not-na, 2023-01-18:
I know that the limitation is not on the `esp-serial-flasher` side, but it only surfaces here, so that's why I opened this issue here.

I'm not quite sure where the limit is coming from - either the ESP32's bootrom or the flash chip. I would guess that it's the former, but can't say for certain without looking at the bootrom source code.
Even if the limit is dependent on the flash chip or exact ESP chip used, it would still be nice to mention that at least 1024 is a safe value and higher values may cause issues.

There seems to be little reason for higher values anyway (doesn't seem to impact performance measurably when comparing between 1024 and 4096), so it could even be considered to just put a hard cap on it and return a meaningful error if the block size is larger than, say, 1024.

dobairoland, 2023-01-23:
>  it would still be nice to mention that at least 1024 is a safe value and higher values may cause issues.

1024 is used in the examples so most of the users will use that by default. Beside that, putting any limit, recommendation or check should be based on some actual and reliable information. We can't have this for every MCU and flash chip. Therefore, we've decided to close this suggestion. Thank you for it and I'm sorry for rejecting it.

closed 2023-01-23 by dobairoland