espressif/esp-serial-flasher #61: flash_begin erase size and esp8266 erase bug?

iKlask, 2023-03-30:
I noticed in function `esp_loader_flash_start` the erase_size is calculated as `erase_size = block_size * blocks_to_write`. However in esptool.py the `erase_size` for the flash_begin command just uses the `image_size`. 

Also, in esptool.py the esp8266 ROM is marked as having an erase bug for flash_begin. There's a work around where the following is used (I translated it to C) for `erase_size` instead:

```
#define FLASH_SECTOR_SIZE 0x1000

uint8_t sectors_per_block = 16;
uint32_t num_sectors = (size + FLASH_SECTOR_SIZE - 1) / FLASH_SECTOR_SIZE;
uint32_t start_sector = offset / FLASH_SECTOR_SIZE;

uint32_t head_sectors = sectors_per_block - (start_sector % sectors_per_block);
if (num_sectors < head_sectors) { head_sectors = num_sectors; }

if (num_sectors < 2 * head_sectors) { return (num_sectors + 1) / 2 * FLASH_SECTOR_SIZE; } 
return (num_sectors - head_sectors) * FLASH_SECTOR_SIZE;
```

Is there a reason this is done differently here vs esptool.py?

DNedic, 2023-03-30:
Thank you for submitting this @iKlask , I will take a look at this.

iKlask, 2023-03-30:
I don't have a working example, as I've never actually run into this bug before. Its just something I noticed when[ flipping through the docs](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/serial-protocol.html#erase-size-bug) and comparing to this project.

I assume the `block_size * blocks_to_write` is to make sure the entire block gets erased for the final block. However its interesting that esptool.py doesn't bother with this, and just sends the exact image size.

DNedic, 2023-03-30:
As preliminary information, it might be because esptool assumes you're using the flasher stub by default. It fixes bugs found in the ROM in the meantime and adds new features that don't exist in older ROMs.

closed 2023-05-29 by espressif-bot