espressif/esp-serial-flasher #63: Flash ESP32-C2 (revision v1.0) error

larryli, 2023-04-20:
Failed to detect size and erase after add C2 v1.0 magic value. See #62

```
DEBUG: Flash size detection failed, falling back to default
Erasing flash failed with error 2.
```

I test esptool.py and check the source, found:

```python
elif args.no_stub:
    print("Enabling default SPI flash mode...")
    # ROM loader doesn't enable flash unless we explicitly do it
    esp.flash_spi_attach(0)
```

https://github.com/espressif/esptool/blob/8f8f50817eb36cadff1301b687789b6c6ebbd71e/esptool/__init__.py#L734-L737

But esp-serial-flasher donot use the empty value of spi_attach:

https://github.com/espressif/esp-serial-flasher/blob/d50cb2b01585cd0f8cfb94e1a16ee58c2272054a/src/esp_loader.c#L115-L117

So I try `err = loader_spi_attach_cmd(0);` 

Everything is OK.

larryli, 2023-04-20:
Donot change HIGHER_BAUDRATE for 26Mhz XTAL.

https://github.com/espressif/esp-serial-flasher/blob/d50cb2b01585cd0f8cfb94e1a16ee58c2272054a/examples/common/example_common.c#L161

Patch:

```c
        err = esp_loader_change_transmission_rate(higher_transmission_rate * 40 / 26);
```

DNedic, 2023-05-24:
This is an issue with a batch of boards using an 26 MHz crystal and the solution would be to do the calculation you provided and pass the calculated baudrate to the port config config (and later recalculate for `esp_loader_change_transmission_rate()` if you want higher speeds). See instructions [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c2/get-started/establish-serial-connection.html#verify-serial-connection).

As it is not inherently a problem with `esp-serial-flasher` and all that is needed to fix it is to set the baudrate correctly in the port config, I believe this should be closed.