espressif/esp-serial-flasher #65: esp_loader_connect fails with ESP-01

gabonator, 2023-04-25:
`esp_loader_connect` successfully detects ESP8266 chip, but following piece of code fails:
```
    if (s_target == ESP8266_CHIP) {
        err = loader_flash_begin_cmd(0, 0, 0, 0, s_target);
    } else {
```

It transmits following packet:
`c0 00 02 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c0`
And esp01 sends back:
`c0 01 02 02 00 01 c1 f0 ff 01 05 c0`

Looking at the [documentation](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/serial-protocol.html) the trailing part `01 05` translates to error: `“Received message is invalid” (parameters or length field is invalid)` Which gives sense, since we are trying to flash 0 bytes starting at address 0.

I couldn't find any documentation on this "probing FLASH_BEGIN packet", how does it work? Why my board refuses this command? (Running on LA104 based on STM32F103)

jtomani, 2023-05-03:
I've encountered this issue with both the ESP-01 and the ESP-12F. Commenting out the call to `loader_flash_begin_cmd` allowed the rest of the flash uploader to work correctly.

jtomani, 2023-05-09:
> I've encountered this issue with both the ESP-01 and the ESP-12F. Commenting out the call to `loader_flash_begin_cmd` allowed the rest of the flash uploader to work correctly.

An update to this is that, while I am indeed able to write flash on the ESP-01 and the ESP-12F, the written flash is failing consistently with a checksum error over the application flash. This may or may not be related to the faulty `loader_flash_begin_cmd` call. I'll post an update once I have a solution.

DNedic, 2023-05-15:
Thank you for investigating this, I will investigate the issue.

think-mcfly, 2023-06-30:
I was having the same issue, but it seemed like the .encrypted parameter in flash_begin_command_t is unnecessary, and for some reason when ESP8266 is the slave device, the field is included in the .size parameter and it is included in the data sent over UART. I removed the .encrypted field from flash_begin_command_t and removed 

uint32_t encryption_size = encryption ? sizeof(uint32_t) : 0;

and subsequent uses of encryption_size in the loader_flash_begin_cmd function, and everything seems to work fine.

DNedic, 2023-06-30:
Thank you for exploring this @think-mcfly . I will explore this next week, on first glance the way we determine whether the encryption field should be true or false is wrong, but I have to test this.

DNedic, 2023-07-07:
I have managed to reproduce the issue, fix is on the way.

closed 2023-07-17 by espressif-bot