espressif/esp-serial-flasher #68: Can't enter boot mode on esp8266

Iceburgino, 2023-06-15:
### Port

ESP

### Target chip

ESP8266

### Hardware Configuration

board - ESP-WROOM-02

### Log output

```plain
DEBUG: Error: 
DEBUG: INVALID_COMMAND
DEBUG:
```

### More Information

When trying to enter boot mode on target esp8266, output of the command `esp_loader_connect` is always this:

DEBUG: Error: 
DEBUG: INVALID_COMMAND
DEBUG: 

The same code and config works for esp32.

think-mcfly, 2023-06-30:
I was having the same issue. Removing the seemingly purposeless .encryption field from the flash_begin_command_t message fixed it for me. See [https://github.com/espressif/esp-serial-flasher/issues/65](https://github.com/espressif/esp-serial-flasher/issues/65)

DNedic, 2023-07-31:
I'm closing this for inactivity

closed 2023-07-31 by DNedic