espressif/esp-serial-flasher #69: Wrong Pinout in Readme

think-mcfly, 2023-06-29:
### Port

ESP

### Target chip

ESP32

### Hardware Configuration

ESP32 --> ESP32

### Log output

```plain
not sure what this is
```

### More Information

Pinout in the readme for ESP32 example does not match the code. Thanks for the repo! This will save me a lot of time!

[image]

[image]

think-mcfly, 2023-06-29:
Actually, looking into the code more, it seems like those are supposed to be the RX/TX pins on the host side (which makes more sense) but now I am really confused because when I made the connections as specified in the README, I always get a timeout error, but when I swap the TX/RX pins, it successfully programs the ESP32, and I verified the hello world program running on the slave ESP32.

DNedic, 2023-06-29:
Hello @think-mcfly , thank you for the submission. Yes you are correct, the table matches host <-> target pins , as this is serial communication, TX of the slave connects to the RX of the host and vice versa. You can clearly see that the host uses for instance pin 4 for tx, and from the table that connects to the RX0 pin of the slave. 

Can you recheck whether you used exactly the connections specified in the README?

DNedic, 2023-06-30:
I'm happy you found the issue, this is why personally I always like to fall back to the documentation of the chip vendor on the first sight of inconsistencies.

closed 2023-06-30 by DNedic