espressif/esp-serial-flasher #73: 增加flasher过程中发生错误后重试的机制

Ruoyu722, 2023-08-24:
### Is your feature request related to a problem?

我在尝试用ESP32S3给另一片ESP32S3烧录的时候 会发生错误
Packet could not be written! Error 2.
发生错误后就会返回失败,这很不方便. 
使用波特率115200 230400时,发生错误的几率不大, 使用波特率460800 921600的时候,出现这个错误的概率就极大. 
我的idf环境为idf v5.0.1 用的是example里的esp32_example工程

### Describe the solution you'd like

我希望能在发生错误时进行重试,比如当前包发送错误时,会重新发送这个包

DNedic, 2023-08-27:
Hello @Ruoyu722 , thank you for your feedback. Can you describe the hardware setup? Since UART is an asynchronous protocol, wire length and electrical noise have an effect on the transmission quality, and 460800 or 921600 are rather high baudrates.

Additionally, we will consider your request for packet retries as a feature request.

Ruoyu722, 2023-08-28:
硬件配置:两块ESP32S3开发板, 中间使用26awg的杜邦线连接. 导线长度在30cm左右
我也认为这跟导线长度和噪声有关, 因为我换短一点的杜邦线的时候确实会成功率高一点. 
在pc端使用 esptool 的时候 460800 921600 波特率时候不会出现失败, 同样的杜邦线连接方式.

DNedic, 2023-08-29:
Yes, 30cm is definitely a bit long for 460800 or 921600. The reason `esptool` works more reliably is because it does have retries for some flash commands.

We are keeping this issue open as a feature request.

closed 2023-10-12 by espressif-bot