espressif/esp-serial-flasher #75: Error 2 on every connection function ESP32 C3 mini Zephyr.

juser19201, 2023-09-04:
### Port

Zephyr

### Target chip

esp32 c3 mini

### Hardware Configuration

Custom board with an MCU running Zephyr OS then the ESP chip is connected along UART3. The boot and en pins at standard GPIO pins. 

### Log output

```plain
uart:~$ Running ESP Flasher from Zephyr

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---
00
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 0                                     0 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07                                      12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55                                      55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 5                                     5 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55                                      55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55                                      55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 5                                     5 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55                                      55 55 c0 Cannot connect to target. Error: 2

uart:~```
```

### More Information

I am running what is pretty much the example code for the zephyr port, my dts and dtsi for the uart and ESP chip are all correct. I can see on my logic analyzer that the data is being sent and read correctly over UART but for some reason the ESP returns 0 and then the code fails? I'm at a loss,

Code: 
`    /* Get UART DTS entry used as flash interface */
    static const struct device *esp_uart_dev = DEVICE_DT_GET(DT_NODELABEL(lpuart3));
    /* Get GPIO pin connected to the ESP's enable pin. */
    static const struct gpio_dt_spec esp_enable_spec = GPIO_DT_SPEC_GET(DT_ALIAS(en), gpios);
    /* Get GPIO pin  connected to the ESP's boot pin. */
    static const struct gpio_dt_spec esp_boot_spec = GPIO_DT_SPEC_GET(DT_ALIAS(boot), gpios);

    uint8_t data[100] = {0};

    #define HIGHER_BAUDRATE 230400
    #define DEFAULT_BAUDRATE 115200

    const loader_zephyr_config_t config = {
            .uart_dev = esp_uart_dev,
            .enable_spec = esp_enable_spec,
            .boot_spec = esp_boot_spec
    };

    printk("Running ESP Flasher from Zephyr\r\n");

    if (!device_is_ready(esp_uart_dev)) {
        printk("ESP UART not ready");
        return 1;
    }

    if (!device_is_ready(esp_boot_spec.port)) {
        printk("ESP boot GPIO not ready");
        return 1;
    }

    if (!device_is_ready(esp_enable_spec.port)) {
        printk("Bluetooth Enable GPIO not ready");
        return 1;
    }

    gpio_pin_configure_dt(&esp_boot_spec, GPIO_OUTPUT_ACTIVE);
    gpio_pin_configure_dt(&esp_enable_spec, GPIO_OUTPUT_INACTIVE);

    if (loader_port_zephyr_init(&config) != ESP_LOADER_SUCCESS) {
        printk("ESP loader init failed");
        return 1;
    }

    static const target_registers_t *s_reg = NULL;
    static target_chip_t s_target = ESP_UNKNOWN_CHIP;

    //get_example_binaries(esp_loader_get_target(), &bin);

    if (connect_to_target(HIGHER_BAUDRATE) == ESP_LOADER_SUCCESS) {
        printk("Success connecting to esp!!\n");

//
//        get_example_binaries(esp_loader_get_target(), &bin);
//
//        flash_binary(bin.boot.data, bin.boot.size, bin.boot.addr);
//        flash_binary(bin.part.data, bin.part.size, bin.part.addr);
//        flash_binary(bin.app.data,  bin.app.size,  bin.app.addr);
    }

    loader_port_change_transmission_rate(DEFAULT_BAUDRATE);

    loader_port_reset_target();`

[image]

DNedic, 2023-09-05:
Hello @juser19201 , thank you for your submission.

Can you provide the exact wiring setup? To me it looks like the second signal in the logic analyzer trace is either `BOOT` or `RESET`, not an UART line. 

Your code seems fine, but you do not need these variables:

```c
static const target_registers_t *s_reg = NULL;
static target_chip_t s_target = ESP_UNKNOWN_CHIP;
```
the first one is a private variable for the loader and the target itself can be obtained with `esp_loader_get_target()`.

juser19201, 2023-09-05:
hiya, thankyou for your response. turns out it was because the ESP had nothing on it and it was booting into USB download mode. Addidtionally IO8 is floating therefore it will not go into serial mode.

I have AT commands working back and forth, was looking into writing a new AT command that handles the update on the ESP itself.

Cheers for your help

juser19201, 2023-09-05:
Thankyou for that, but we'd like something without internet access. The thinking is that if something goes wrong, we need a local solution. Additionally, the issue with simply flashing the slave ESP is that we can not use the serial to flash it. The only way to flash it is via USB which is why we need something AT related.

If i implement my own AT command to take a file over UART, can i flash the ESP with a file thats loaded onto the ESP?

DNedic, 2023-09-05:
You should be able to do that, yes, but I don't see why you can't reflash the slave the same way you flashed it initially from your explanation. 

Anyway, I'm closing this issue as completed since the original issue is solved, for additional help feel free to open another issue and explain the setup in detail.

Additionally, you or your company can contact our [Circuit Schematic & PCB Design Review](Circuit Schematic &
PCB Design Review) service for your hardware design.

closed 2023-09-05 by DNedic