espressif/esp-serial-flasher #76: ESP32C3 flash end command failed

LINGlong-zi, 2023-09-07:
### Port

ESP32

### Target chip

ESP32C3

### Hardware Configuration

** Master board: Olimex esp32-poe board
** Target board: Seeed Xiao esp32c3 board
** Baudrate: 115200*8
** Connection:
UART input and output are set correctly between the two boards, but I'm not connecting the reset and chip_en pins. Instead, I'm using a software method for the esp32c3 to enter the download mode. 

** Problem description: 
I'm using the example `esp32_example`, and I works fine for me right now. The problem is that I want the esp32c3 board to reboot after the firmware is successfully flashed. Thus I call the function `esp_loader_flash_finish(false)` after the flash tasks are completed. However, this command failed and I cannot figure out why.

### Log output

```plain
** Output at esp32-poe(master) board without SERIAL_FLASHER_DEBUG_TRACE enabled
I (33) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (33) boot: compile time Sep  7 2023 14:16:44
I (33) boot: Multicore bootloader
I (38) boot: chip revision: v3.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00177000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1117e0h (1120224) map
I (337) esp_image: segment 1: paddr=00121808 vaddr=3ffb0000 size=02148h (  8520) load
I (340) esp_image: segment 2: paddr=00123958 vaddr=40080000 size=0c6c0h ( 50880) load
I (356) esp_image: segment 3: paddr=00130020 vaddr=400d0020 size=18aach (101036) map
I (377) esp_image: segment 4: paddr=00148ad4 vaddr=4008c6c0 size=00658h (  1624) load
I (385) boot: Loaded app from partition at offset 0x10000
I (385) boot: Disabling RNG early entropy source...
I (397) cpu_start: Multicore app
I (398) cpu_start: Pro cpu up.
I (398) cpu_start: Starting app cpu, entry point is 0x400811c8
0x400811c8: call_start_cpu1 at /Users/guoxiaolinglong/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (416) cpu_start: Pro cpu start user code
I (416) cpu_start: cpu freq: 160000000 Hz
I (416) cpu_start: Application information:
I (421) cpu_start: Project name:     esp-serial-flasher
I (426) cpu_start: App version:      aa5137a-dirty
I (432) cpu_start: Compile time:     Sep  7 2023 16:26:47
I (438) cpu_start: ELF file SHA256:  e6c948df31351296...
I (444) cpu_start: ESP-IDF:          v5.1-dirty
I (449) cpu_start: Min chip rev:     v0.0
I (454) cpu_start: Max chip rev:     v3.99 
I (459) cpu_start: Chip rev:         v3.0
I (464) heap_init: Initializing. RAM available for dynamic allocation:
I (471) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (477) heap_init: At 3FFB2E30 len 0002D1D0 (180 KiB): DRAM
I (483) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (489) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (496) heap_init: At 4008CD18 len 000132E8 (76 KiB): IRAM
I (503) spi_flash: detected chip: generic
I (507) spi_flash: flash io: dio
I (511) app_start: Starting scheduler on CPU0
I (516) app_start: Starting scheduler on CPU1
I (516) main_task: Started on CPU0
I (526) main_task: Calling app_main()
I (526) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (536) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
sent
loader_port_enter_bootloader
loader_initialize_conn: 0
loader_detect_chip: 0
Connected to target
Transmission rate changed changed
I (4306) serial_flasher: Loading bootloader...
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
Flash verified
I (5696) serial_flasher: Loading partition table...
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
Flash verified
I (5956) serial_flasher: Loading app...
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
Flash verified
I (15006) serial_flasher: Done!
DEBUG: Error: 
DEBUG: COMMAND_FAILED
DEBUG: 

I (15016) main_task: Returned from app_main()

** Output at esp32-poe(master) board with SERIAL_FLASHER_DEBUG_TRACE enabled
Progress: 100 %20 40 16 00 00 00 36 31 c0 
Finished programming

--- WRITE ---
c0 00 13 10 00 00 00 00 00 00 00 01 00 70 0d 02 00 00 00 00 00 00 00 00 00 c0 
--- READ ---
c0 01 13 24 00 20 40 16 00 32 66 66 39 39 39 31 35 30 34 38 63 61 33 64 39 32 36 33 35 33 32 64 65 32 39 32 63 61 35 62 62 00 00 00 00 c0 Flash verified
I (52316) serial_flasher: Done!

--- WRITE ---
c0 00 04 04 00 00 00 00 00 01 00 00 00 c0 
--- READ ---
c0 01 04 04 00 20 40 16 00 01 06 66 39 c0 DEBUG: Error: 
DEBUG: COMMAND_FAILED
DEBUG: 

** Output at esp32c3(target) board

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x4 (DOWNLOAD(USB/UART0/1))
Saved PC:0x400462e2
0x400462e2: ets_efuse_get_wp_pad in ROM

waiting for download
user code done
Guru Meditation Error: Core 0 panic'ed (Illegal instruction)
Core 0 register dump:
Stack dump detected
PC      : 0x00000000  RA      : 0x00000000  SP      : 0x3fcde710  GP      : 0x00000000
TP      : 0x00000000  T0      : 0x4004b94c  T1      : 0x00000000  T2      : 0x3a568130
0x4004b94c: uart_baudrate_detect in ROM

S0      : 0x00000000  S1      : 0x00000000  A0      : 0x00000000  A1      : 0x0000000a
A2      : 0x3fcde6e4  A3      : 0x6000001c  A4      : 0x00000000  A5      : 0x00000000
A6      : 0xd9a38c04  A7      : 0xd4d64705  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0xc4ef9070  T4      : 0x0243bfbf  T5      : 0x534742b2  T6      : 0x39d774f2
MSTATUS : 0x00001881  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  INTLEVEL: 0x00000001
```

LINGlong-zi, 2023-09-07:
In the log part three logs are provided:
Output at esp32-poe(master) board without SERIAL_FLASHER_DEBUG_TRACE enabled
Output at esp32-poe(master) board with SERIAL_FLASHER_DEBUG_TRACE enabled
Output at esp32c3(target) board

DNedic, 2023-09-11:
Hello @LINGlong-zi , thank you for your submission, we will try to reproduce your issue and fix it as soon as possible.

LINGlong-zi, 2023-09-12:
Thanks a lot! I still haven't solved the problem. Really looking forward to your result.

LINGlong-zi, 2023-09-12:
> Hello @LINGlong-zi , thank you for your submission, we will try to reproduce your issue and fix it as soon as possible.
Thanks a lot! I still haven't solved the problem. Really looking forward to your result.

Dzarda7, 2025-06-23:
Hi @LINGlong-zi, sorry for late response. For chips other than ESP8266 `FLASH_END` command has no point (it still should not fail), the reset in the command does nothing for the chips. You do not need to call that and flashing will be successful.