espressif/esp-serial-flasher #78: Zephyr port running on rpi pico buil issue

erhan-namli, 2023-09-14:
### Port

Zephyr

### Target chip

ESP32-S3

### Hardware Configuration

The connections are okay, I'm having trouble after the "west build -b rpi_pico" command 

### Log output

```plain
[1/13] Performing build step for 'second_stage_bootloader'
ninja: no work to do.
[4/11] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj
ccache /home/erhan/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DSERIAL_FLASHER_INTERFACE_UART -D_ANSI_SOURCE -D_FORTIFY_SOURCE=1 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/../common -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/../binaries -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/../../port -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/../../src -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/../../include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/../../private_include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/build/zephyr/include/generated -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/soc/arm/rpi_pico/rp2 -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/lib/libc/newlib/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/../../zephyr/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/.. -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/source/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/source/include/platform -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/source/compiler -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/generate/zephyr -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/source/tools/acpiexec -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/lib/acpica/source/tools/acpidump -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/cmsis/CMSIS/Core/Include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/modules/cmsis/. -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_base/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_clocks/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_watchdog/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_xosc/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_pll/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_irq/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_sync/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_timer/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_resets/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/pico_bootrom/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2040/hardware_regs/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2040/hardware_structs/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/common/pico_base/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/pico_platform/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/modules/hal_rpi_pico -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_gpio/include -I/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/modules/hal/rpi_pico/src/rp2_common/hardware_uart/include -fno-strict-aliasing -Os -imacros /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/build/zephyr/include/generated/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m0plus -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/erhan/zephyr-sdk-0.16.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/erhan/playground/esp-serial-flasher/examples/zephyr_example=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/erhan/playground/esp-serial-flasher/examples/zephyr_example=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/src/main.c
In file included from /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/toolchain/gcc.h:92,
                 from /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/toolchain.h:50,
                 from /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/kernel_includes.h:19,
                 from /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/kernel.h:17,
                 from /home/erhan/playground/esp-serial-flasher/examples/zephyr_example/src/main.c:12:
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/device.h:85:41: error: '__device_dts_ord_DT_N_ALIAS_uart1_ORD' undeclared here (not in a function)
   85 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/toolchain/common.h:132:26: note:
 in definition of macro '_DO_CONCAT'
  132 | #define _DO_CONCAT(x, y) x ## y
      |                          ^
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/device.h:85:33: note: in expansion of macro '_CONCAT'
   85 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/device.h:211:37: note: in expansion of macro 'DEVICE_NAME_GET'
  211 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_ID(node_id))
      |                                     ^~~~~~~~~~~~~~~
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/device.h:228:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
  228 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
      |                                  ^~~~~~~~~~~~~~~~~~
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/src/main.c:28:44: note: in expansion of macro 'DEVICE_DT_GET'
   28 | static const struct device *esp_uart_dev = DEVICE_DT_GET(DT_ALIAS(uart1));
      |                                            ^~~~~~~~~~~~~
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/zephyr/include/zephyr/device.h:85:41: error: '__device_dts_ord_DT_N_ALIAS_en_P_gpios_IDX_0_PH_ORD' undeclared here (not in a function)
   85 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
[... cut ...]
      |                                                                   ^~~~~~~~
/home/erhan/playground/esp-serial-flasher/examples/zephyr_example/src/main.c:34:6: warning: return type of 'main' is not 'int' [-Wmain]
   34 | void main(void)
      |      ^~~~
ninja: build stopped: subcommand failed.
```

### More Information

I think the problem is about my Zephyr dev environment, I'm trying to build the Zephyr port of the esp serial flasher to the Raspberry Pi Pico board and I'm trying to flash the esp32 s3 board with RPI Pico.

DNedic, 2023-09-19:
Hello @erhan-namli , the issue you are facing is most probably the lack of, or an improper device tree `.overlay` file.

You can read about device trees and `.overlay` files [here](https://docs.zephyrproject.org/latest/build/dts/howtos.html). 

If this is not the issue, please provide more information on your setup such as the exact steps you took to build the example.

DNedic, 2023-09-19:
Here's an example `rpi_pico.overlay` I created just to get the example to build:
```dts
/ {
	aliases {
		en = &en_button;
		boot = &boot_button;
        uart1 = &uart0;
	};

	gpio_keys {
		compatible = "gpio-keys";
		en_button: en_button {
			gpios = <&gpio0 6 (GPIO_PULL_UP | GPIO_ACTIVE_HIGH)>;
		};
		boot_button: boot_button {
			gpios = <&gpio0 7 (GPIO_PULL_UP | GPIO_ACTIVE_HIGH)>;
		};
	};
};
```

obviously the uart1 = &uart0; mapping is counterintuitive, we will try giving this a better alias.

erhan-namli, 2023-09-21:
Can you give me a straightforward build pipeline for Raspberry Pi Pico within your main repository? I'm a bit confused with source directories and Zephyr project build steps.

DNedic, 2023-09-21:
The only steps missing from our example README are `west init` and `west update`, which are used to create a zephyr [workspace](https://docs.zephyrproject.org/latest/develop/application/index.html#zephyr-workspace-application). We will add these steps to the README later.

You need to create the `.overlay` inside the `boards` directory and proceed to build the project for `rpi_pico`.

Other than that, just follow the example documentation.

closed 2023-10-02 by dobairoland