espressif/esp-serial-flasher #79: Can't flash above 115200 baud

LeedanJohnson, 2023-10-09:
### Port

ESP

### Target chip

ESP32

### Hardware Configuration

I'm using a minimalist custom board with only the bare minimum (voltage regulator, capacitors, etc.) there are no peripherals or anything else attached to the board. I've checked that power is smooth and sufficient, and that the issue is consistent across multiple boards as flasher and flashee.

### Log output

```plain
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
65 74 73 20 4a 75 6e 20 20 38 20 32 30 31 36 20 30 30 3a 32 32 3a 35 37 0d 0a 0d 0a 72 73 74 3a 30 78 31 20 28 50 4f 57 45 52 4f 4e 5f 52 45 53 45 54 29 2c 62 6f 6f 74 3a 30 78 33 20 28 44 4f 57 4e 4c 4f 41 44 5f 42 4f 4f 54 28 55 41 52 54 30 2f 55 41 52 54 31 2f 53 44 49 4f 5f 52 45 49 5f 52 45 4f 5f 56 32 29 29 0d 0a 77 61 69 74 69 6e 67 20 66 6f 72 20 64 6f 77 6e 6c 6f 61 64 0d 0a 
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 
--- WRITE ---
c0 00 0a 04 00 00 00 00 00 00 10 00 40 c0 
--- READ ---
c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 c0 01 08 04 00 07 12 55 55 00 00 00 00 c0 c0 01 0a 04 00 83 1d f0 00 00 00 00 00 c0 
--- WRITE ---
c0 00 0a 04 00 00 00 00 00 14 a0 f5 3f c0 
--- READ ---
c0 01 0a 04 00 00 00 00 00 00 00 00 00 c0 
--- WRITE ---
c0 00 0a 04 00 00 00 00 00 0c a0 f5 3f c0 
--- READ ---
c0 01 0a 04 00 00 a2 00 00 00 00 00 00 c0 
--- WRITE ---
c0 00 0d 08 00 00 00 00 00 00 00 00 00 00 00 00 00 c0 
--- READ ---
c0 01 0d 04 00 00 a2 00 00 00 00 00 00 c0 Connected to target

--- WRITE ---
c0 00 0f 08 00 00 00 00 00 00 84 03 00 00 00 00 00 c0 
--- READ ---
c0 01 0f 04 00 00 a2 00 00 00 00 00 00 c0 Transmission rate changed changed
[I][main.cpp:60] setup(): Loading app...
Erasing flash (this may take a while)...
DEBUG: Flash size detection failed, falling back to default

--- WRITE ---
c0 00 02 10 00 00 00 00 00 20 02 04 00 01 01 00 00 00 04 00 00 00 00 01 00 c0 
--- READ ---
60 c0 01 02 04 00 00 a2 00 00 00 00 00 00 c0 Start programming

Packet could not be written! Error 2.
```

### More Information

 I see esptool can flash at 460800 or higher, and the esp32 uart should be capable of high-speed transmissions, but invariably using a higher baud rate fails. Interestingly I can mitigate the issue to some extent by adding delays between each packet, but this fails to improve overall throughput.

DNedic, 2023-10-10:
Hello @LeedanJohnson , thank you for your submission. What is your wiring setup? Due to electrical noise, and the asynchronous nature of UART wire length can be a factor in whether you can achieve higher speeds.

It would be beneficial to see any logic analyzer traces if you have them as well.

LeedanJohnson, 2023-10-10:
I currently have dupont connector jumper wires connecting the two devices. Using the same wiring, if I substitute in a usb->ttl board like the ch340, the board flashes perfectly at 460800 baud or higher, so I know it's not the wiring.

I don't have logic analyzer traces, but could get them if it proved necessary.

I've been investigating how esptool handles flashing, and I see that it first uploads a stub, then performs the high-speed flashing. Is it confirmed that the normal bootloader ROM can handle speeds above 115200, or is that only possible using a stub?

DNedic, 2023-10-10:
The ROM can handle 460800 or higher, the difference is in the fact that `esptool` retries failed flashing packets, while `esp-serial-flasher` does not. We aim to bring this functionality to `esp-serial-flasher` as well. 

Until then, we should confirm whether wire length is indeed the problem, please try shorter wires and provide a logic analyzer trace.

DNedic, 2023-10-12:
We've added block write retries, please check if this solves your problem.

dobairoland, 2023-10-16:
> We've added block write retries, please check if this solves your problem.

Added by https://github.com/espressif/esp-serial-flasher/commit/c1a99b4accf6c246dfe41578fa5b437950dad1e6 and released in version 0.4.0.

DNedic, 2023-11-01:
The timeout happens inside `esp-serial-flasher` itself, the ROM does not have the concept of timeout. However, the default timeout is 1s which I find very hard to imagine happens naturally. It is possible that some bytes are lost due to software issues or physical connection issues.

Please increase the `DEFAULT_TIMEOUT` variable for the time being and recheck. If that doesn't work, submit a logic analyzer trace so we can be sure, as right now debug printing can only print whole packets when they are already parsed.

dobairoland, 2023-11-13:
I'm closing this for inactivity. We can reopen if additional information is given....

closed 2023-11-13 by dobairoland