espressif/esp-serial-flasher #80: ESP32-C6 with seemingly different efuse bits

pieterjanbuntinx, 2023-10-23:
### Port

ESP

### Target chip

ESP32-C6

### Hardware Configuration

I am using a custom board with an ESP32-S3 and a ESP32-C6. These are connected like in the ESP_ZIGBEE_GATEWAY example: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway 
I am also using this code from this example for testing.

Pinout:

- RESET: 17
- BOOT: 18
- TX: 5
- RX: 4

Baudrate: 115200

### Log output

```plain
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037c4ce
0x4037c4ce: esp_cpu_wait_for_intr at C:/Users/Pieter-JanBuntinx/esp_v5.1.1/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (27) boot: compile time Oct 20 2023 18:12:11
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (72) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (79) boot:  2 nvs              WiFi data        01 02 0000c000 00004000
I (87) boot:  3 nvs_prov         WiFi data        01 02 00010000 00004000
I (94) boot:  4 nvs_keys         NVS keys         01 04 00014000 00001000
I (102) boot:  5 zb_storage       Unknown data     01 81 00015000 00020000
I (109) boot:  6 zb_fct           Unknown data     01 81 00035000 00000400
I (117) boot:  7 rcp_fw           Unknown data     01 82 00036000 000af000
I (124) boot:  8 ota_0            OTA app          00 10 000f0000 0037a000
I (132) boot:  9 ota_1            OTA app          00 11 00470000 0037a000
I (139) boot: End of partition table
I (144) esp_image: segment 0: paddr=000f0020 vaddr=3c110020 size=3e168h (254312) map
I (198) esp_image: segment 1: paddr=0012e190 vaddr=3fc99e00 size=01e88h (  7816) load
I (200) esp_image: segment 2: paddr=00130020 vaddr=42000020 size=104204h (1065476) map
I (395) esp_image: segment 3: paddr=0023422c vaddr=3fc9bc88 size=02780h ( 10112) load
I (397) esp_image: segment 4: paddr=002369b4 vaddr=40374000 size=15d10h ( 89360) load
I (429) boot: Loaded app from partition at offset 0xf0000
I (430) boot: Disabling RNG early entropy source...
I (441) cpu_start: Multicore app
I (441) cpu_start: Pro cpu up.
I (441) cpu_start: Starting app cpu, entry point is 0x403756b0
0x403756b0: call_start_cpu1 at C:/Users/Pieter-JanBuntinx/esp_v5.1.1/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
[... 263 lines cut ...]

0x420415f0: zb_mlme_dev_reset_conf at ??:?

0x4201b4f6: zb_sched_loop_iteration at ??:?

0x4203417b: zboss_main_loop_iteration at ??:?

0x420103ff: esp_zb_main_loop_iteration at ??:?

0x4200a38e: esp_zb_task at C:/GIT/wilms-gateway-fw/main/main.c:379 (discriminator 2)

0x40381451: vPortTaskWrapper at C:/Users/Pieter-JanBuntinx/esp_v5.1.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 9a77bc40e720876b

Rebooting...
```

### More Information

This setup does work with an ESP32-C6-DevKitC-1 V1.1 (it has the sticker ESP32-C6 Sample Chip V0.0) and an ESP32-S3-DevKitC-1 V1.0 connected in the same way as my custom board. The change below to `esp_targets.c` on line 278 makes it work on both my custom board and the setup with the Dev boards. My guess is that the serial flasher support was only tested on the prerelease ESP32-C6 chips and not the release chips that I have.

```
if (pins == 0 || pins == 0xffffffff || pins == 0xFFFE) {
```

DNedic, 2023-10-26:
Thank you for your submission and detailed investigation @pieterjanbuntinx , I will investigate the issue.

closed 2023-10-31 by espressif-bot

DNedic, 2023-10-31:
This one was interesting to find, because the extended MAC is now in the place of one of the SPI config efuse registers, and the test could pass depending on the MAC of the device tested.

pieterjanbuntinx, 2023-10-31:
Okay, good to know! So the "fix" I suggested wouldn't have worked on all ESP32-C6 chips. 

The new version of the component works for me now :)