espressif/esp-serial-flasher #83: DEFLATE_ERROR on compressed flash download

CMAICEpower, 2023-11-20:
Hi, I am attempting to flash an Esp32 via compressed flash download. I have got it working with the normal flash download, however when attempting to use the compressed flash download I get the following error:
"DEFLATE_ERROR".

I can't seem to find any detailed documentation on how to use the compressed flash download other than it functioning the same as Ram download & Flash download, and the data being compressed using gzip deflate algorithm.

As of now, I have attempted to flash the esp32 with gzip compressed binaries as well as uncompressed binaries, and changing the commands to use FLASH_DELF_BEGIN, FLASH_DELF_DATA & FLASH_DELF_END instead of the flash download equivalents. Either method results in the error "DEFLATE_ERROR".

Any suggestions on what I might be doing wrong?

DNedic, 2023-11-21:
Hello @CMAICEpower , we currently do not support compressed flash download in `esp-serial-flasher`. We will explore this idea in the future and will keep this issue open as a feature request.

In the meantime, feel free to take a look at [this](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html#writing-data) documentation page on the serial protocol and the [esptool source code](https://github.com/espressif/esptool) if you want to either create a quick proof of concept or contribute to the project.

kerms, 2025-12-24:
I put an initial implementation for flashing pre-compressed binaries (host sends a zlib/deflate-compressed stream) using the ROM protocol FLASH_DEFL_* commands. Draft PR -> https://github.com/espressif/esp-serial-flasher/pull/165

To align with expectations for this feature: is on-the-fly compression (compressing during transfer) considered necessary/desirable, or is supporting pre-compressed images the intended scope?

If on-the-fly compression is in scope, is there a preferred host-side compression library (e.g. zlib / miniz / libdeflate) or any constraints to keep in mind (portability, licensing, dependency policy)?

kerms, 2025-12-25:
Thanks for the feedback — I’m aligned with that approach.

One practical note about on-the-fly compression with the ROM protocol: `FLASH_DEFL_BEGIN` requires the **number of data packets** up front (along with uncompressed size, packet size, and flash offset). If we compress while streaming, we don’t know the final packet count until compression is complete.
-> That suggests either a two-pass approach (compress to a buffer/file first, then send) or some protocol-side workaround if available.

Reference: https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/serial-protocol.html#supported-by-stub-loader-and-rom-loader

Also, I’ve completed my Draft PR and marked it **Ready for review**.

Dzarda7, 2025-12-26:
Yes, that is true, I believe no one thought about this when implementing the command, because esptool does it exactly like this. Good thing is that I believe at least no stub checks this value so it can be just random (even though it is not documented anywhere), so if this feature will be added, it can be stated that the packet number is not verified.

Thanks for the PR, I will take a look soon. From the first check it looked very good, thanks.