espressif/esp-serial-flasher #84: Read flash function

rnadyrshin, 2023-11-20:
### Is your feature request related to a problem?

The esp-serial-flasher library flashes very well. 
But sometimes i need to read a flash in the field with a small device, without a computer. Reading support would be very helpful

### Describe the solution you'd like

Read function in esp-serial-flasher or other c/c++ library

DNedic, 2023-11-21:
Hello @rnadyrshin , this is a [stub-only feature](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/serial-protocol.html#supported-by-stub-loader-only). We will look into adding support for the flasher stub and implementing those features, this issue will be kept as a feature request in the meantime.

DNedic, 2024-01-06:
It turns out that there is indeed a command that can be used to read memory in the ROM, we will explore adding this feature.

closed 2024-11-25 by espressif-bot