espressif/esp-serial-flasher #85: ESP32-C3 - INVALID_COMMAND while flashing sometimes

caipiblack, 2023-11-23:
### Port

Linux/Raspberry

### Target chip

ESP32-C3

### Hardware Configuration

I am running the software on UBUNTU and the computer is plugged to a ESP32-C3-DevKitM-1

### Log output

```plain
(incomplete logs because too heavy for github, see attached files)

Progress: 97 %
--- WRITE ---
c0 00 03 10 04 6c 00 00 00 00 04 00 00 25 05 00 00 00 00 00 00 00 00 00 00 97 70 c7 ff e7 80 40 f0 1c 5c 85 07 1c dc 1c 50 be 99 5c 58 23 20 34 03 b3 89 37 41 9d 47 63 e4 37 01 1c 58 1c d0 18 50 5c 54 63 16 f7 00 1c 44 93 e7 47 00 1c c4 58 d0 f2 40 62 44 d2 44 42 49 b2 49 22 4a 92 4a 05 61 82 80 01 11 26 ca ae 84 b2 85 06 ce 22 cc 32 c6 2a 84 ef f0 bf a8 32 46 a6 85 97 70 c7 ff e7 80 20 ea aa 85 22 85 62 44 f2 40 d2 44 05 61 6f f0 3f b3 1d 71 a2 cc a6 ca ce c6 c2 84 d2 c4 d6 c2 da db dc 5e de 62 dc 66 da 6a d8 6e d6 2a 84 86 ce ca c8 2e 8a b2 8a b6 8b 3a 8c 3e c6 ef 20 a0 18 33 0d 95 00 a6 89 13 0b 04 09 85 4c fd 5d ce 85 5a 85 ef 10 50 03 63 03 95 03 01 49 f6 40 66 44 4a 85 d6 44 46 49 b6 49 26 4a 96 4a 06 4b f2 5b 62 5c d2 5c 42 5d b2 5d 25 61 82 80 ef 30 a0 44 22 85 ef f0 af e5 2a 89 63 12 95 0b 1c 44 58 48 de 86 89 8b b1 cf 32 46 81 45 22 85 02 97 1c 44 23 20 aa 00 85 8b bd c7 63 86 0a 06 63 04 0c 06 72 47 85 47 63 1e f7 04 5c 48 e2 86 01 46 6c 08 22 85 82 97 23 a0 aa 00 83 27 0a 00 63 65 f5 02 b7 86 11 3c 37 86 11 3c 93 86 86 2e 13 06 06 e2 93 05 d0 31 37 85 11 3c 13 05 45 0c ef 60 db dc 19 01 46 6c 08 5d b7 f2 47 91 cf b7 86 11 3c 37 86 11 3c 93 86 c6 2f 13 06 06 e2 93 05 e0 31 d9 bf 23 a0 0a 00 00 5c ef 30 a0 3e e3 51 80 f4 81 46 01 46 81 45 5a 85 ef 10 40 19 0d bf 63 86 b4 01 ef 20 60 09 b3 09 ad 40 ef 30 80 3c e3 f9 34 f1 29 bf 39 71 06 de 22 dc 26 da 4e d6 52 d4 2a 84 3a 8a ae 89 b2 84 36 c6 3e c4 4a d8 ef 30 a0 36 22 85 ef f0 af d7 05 47 63 1b e5 0a 18 44 2a 89 03 28 44 01 09 8b b2 46 29 cf a2 47 81 45 3e 86 22 85 02 98 1c 44 23 a0 a9 00 85 8b ad c7 a5 c4 63 03 0a 06 72 47 85 47 63 1d f7 04 5c 48 d2 86 01 46 6c 08 22 85 82 97 88 db dc 83 a7 09 00 63 65 f5 02 b7 86 11 3c 37 86 11 3c 93 86 86 2e 13 06 46 e3 93 05 30 35 37 85 11 3c 13 05 45 0c ef 60 a0 0b 01 46 6c 08 6d b7 f2 47 91 cf b7 86 11 3c 37 86 11 3c 93 86 c6 2f 13 06 46 e3 93 05 40 35 d9 bf 23 a0 04 00 04 5c b3 24 90 00 ef 30 40 30 91 c4 81 45 13 05 04 09 ef 10 20 45 f2 50 62 54 4a 85 d2 54 42 59 b2 59 22 5a 21 61 82 80 81 44 01 49 e9 bf 1c 45 91 8b 9d eb 1c 51 58 55 63 fd e7 00 b3 07 f7 40 e1 17 63 c1 07 02 58 41 63 73 f7 00 ba 87 3e 85 82 80 54 59 10 59 b3 87 f6 40 11 8f e3 f2 e7 fe ba 87 f9 bf 81 47 e5 b7 01 45 82 80 1c 45 91 8b 9d ef 14 51 10 59 58 55 63 9e c6 00 63 9c e6 00 1c 41 e1 17 63 c3 07 02 58 41 63 73 f7 00 ba 87 3e 85 82 80 63 f5 e6 00 b3 07 d7 40 dd b7 5c 59 11 8f 95 8f ba 97 c1 17 f1 bf 81 47 d5 b7 01 45 82 80 1c 45 91 8b 91 eb 5c 55 18 51 99 8f 63 44 f0 00 18 41 ba 97 3e 85 82 80 01 45 82 80 1c 59 41 11 06 c6 63 f2 f5 02 b7 86 11 3c 37 86 11 3c 93 86 06 32 13 06 86 e0 93 05 d0 29 37 85 11 3c 13 05 45 0c ef 50 f0 79 5c 59 63 ed f5 00 b7 86 11 3c 37 86 11 3c 93 86 c6 34 13 06 86 e0 93 05 e0 29 e9 bf 1c 55 5c d5 1c 45 13 f7 47 00 19 c3 ed 9b 1c c5 b2 40 41 01 82 80 18 51 1c 59 63 65 f7 00 5c 59 63 64 f7 02 b7 86 11 3c 37 86 11 3c 37 85 11 3c 41 11 93 86 86 f6 13 06 06 d5 93 05 70 14 13 05 45 0c 06 c6 ef 50 b0 73 5c 55 63 18 f7 00 08 45 09 81 13 45 15 00 05 89 82 80 63 78 f7 00 99 8f b3 b7 b7 00 13 c5 17 00 82 80 1d 8f 1c 41 99 8f 33 b5 b7 00 13 45 15 00 82 80 1c 45 91 8b 8d ef 18 51 5c 55 99 8f 18 41 63 43 f0 00 ba 97 c0 
--- READ ---
Progress: 97 %
--- WRITE ---
c0 00 03 10 04 6c 00 00 00 00 04 00 00 26 05 00 00 00 00 00 00 00 00 00 00 3e 85 63 75 f7 02 b7 86 11 3c 37 86 11 3c 37 85 11 3c 41 11 93 86 06 ff 13 06 86 d2 93 05 40 12 13 05 45 0c 06 c6 ef 50 b0 6c 01 45 82 80 01 11 06 ce 22 cc 26 ca 4a c8 4e c6 0d e1 b7 86 11 3c 37 86 11 3c 93 86 06 56 13 06 06 e5 93 05 a0 36 37 85 11 3c 13 05 45 0c ef 50 90 69 89 47 2e 89 63 fd b7 00 b7 86 11 3c 37 86 11 3c 93 86 06 57 13 06 06 e5 93 05 b0 36 e1 bf aa 84 63 85 f5 00 93 04 35 00 f1 98 93 05 db dc 0e 05 45 ef 50 d0 7e 2a 84 26 85 ef 50 10 79 aa 89 11 db dc 05 e1 22 85 ef 50 d0 78 4e 85 ef 50 70 78 01 44 22 85 f2 40 62 44 d2 44 42 49 b2 49 05 61 82 80 93 06 c4 03 0d 47 01 46 81 45 05 45 ef 00 70 47 93 06 04 09 0d 47 01 46 81 45 05 45 ef 00 70 46 ce 86 22 86 ca 85 26 85 ef f0 1f 8a d1 b7 39 71 06 de 22 dc 26 da 4a d8 4e d6 52 d4 56 d2 5a d0 5e ce 62 cc 66 ca 0d e1 b7 86 11 3c 37 86 11 3c 93 86 86 5b 13 06 46 e6 93 05 30 40 37 85 11 3c 13 05 45 0c ef 50 10 5d 2a 84 2e 8b b2 84 b6 89 89 ed 19 ce b7 86 11 3c 37 86 11 3c 93 86 06 59 13 06 46 e6 93 05 40 40 d1 bf 5c 41 63 ee c7 06 1c 44 89 8b 99 c3 05 49 ad c8 ef 10 90 4d 33 0c 35 01 4e 8a 93 0a c4 03 85 4b fd 5c d2 85 56 85 ef 10 20 38 63 1a 75 05 ef 20 b0 7b 5c 44 a6 85 22 85 82 97 2a 89 63 17 75 03 1c 48 26 86 da 85 22 85 82 97 22 85 ef f0 bf e5 2a c6 ef 20 10 7d 32 45 31 e1 81 46 01 46 81 45 13 05 04 09 ef 00 90 57 21 a8 63 86 99 01 ef 10 b0 47 33 0a ac 40 ef 20 d0 7a e3 f4 49 fb 01 49 f2 50 62 54 4a 85 d2 54 42 59 b2 59 22 5a 92 5a 02 5b f2 4b 62 4c d2 4c 21 61 82 80 81 46 01 46 81 45 56 85 ef 00 90 53 4d bf 79 71 06 d6 22 d4 26 d2 4a d0 4e ce 0d e1 b7 86 11 3c 37 86 11 3c 93 86 86 5b 13 06 46 e7 93 05 00 44 37 85 11 3c 13 05 45 0c ef 50 70 4d 2a 84 ae 89 36 89 89 ed 05 ca b7 86 11 3c 37 86 11 3c 93 86 06 59 13 06 46 e7 93 05 10 44 d9 bf 5c 41 63 fb c7 00 81 44 b2 50 22 54 26 85 02 59 92 54 f2 49 45 61 82 80 1c 44 89 8b 99 c3 85 44 65 d6 32 c6 ef 20 d0 6c 32 46 5c 44 22 85 b2 85 82 97 85 47 aa 84 32 46 63 1d f5 02 1c 48 ce 85 22 85 82 97 22 85 ef f0 9f d6 b3 39 a0 00 ef 20 d0 6d 63 87 09 00 ca 85 13 05 c4 03 ef 10 80 02 85 47 e3 92 f4 fa ca 85 13 05 04 09 ef 10 80 01 61 bf 81 49 81 44 e1 bf 01 11 06 ce 22 cc 0d e1 b7 86 11 3c 37 86 11 3c 37 85 11 3c 93 86 86 5b 13 06 c6 e8 93 05 70 46 13 05 45 0c ef 50 70 41 32 88 81 47 01 47 2e 84 74 00 01 46 2c 00 ef f0 ff 98 05 47 81 47 63 16 e5 00 19 db dc b2 47 1c db dc a2 47 f2 40 62 44 3e 85 05 61 82 80 01 11 06 ce 22 cc 0d e1 b7 86 11 3c 37 86 11 3c 37 85 11 3c 93 86 86 5b 13 06 06 ea 93 05 a0 47 13 05 45 0c ef 50 30 3c 81 47 01 47 2e 84 74 00 01 46 2c 00 ef f0 df a6 05 47 81 47 63 16 e5 00 19 db dc b2 47 1c db dc a2 47 f2 40 62 44 3e 85 05 61 82 80 01 11 06 ce 22 cc 0d e1 b7 86 11 3c 37 86 11 3c 93 86 86 5b 13 06 c6 eb 93 05 b0 50 37 85 11 3c 13 05 45 0c ef 50 10 37 81 ed b7 86 11 3c 37 86 11 3c 93 86 06 65 13 06 c6 eb 93 05 db dc 50 f9 bf 2a 84 2e c6 ef 20 f0 58 b2 45 1c 4c 22 85 82 97 ef 20 d0 5b 13 05 c4 03 62 44 f2 40 81 46 01 46 81 45 05 61 6f 00 30 36 01 11 06 ce 22 cc 0d e1 b7 86 11 3c 37 86 11 3c 93 86 86 5b 13 06 46 ed 93 05 70 51 37 85 11 3c 13 05 45 0c ef 50 90 30 81 ed b7 86 11 3c 37 86 11 3c 93 86 06 65 13 06 46 ed 93 05 c0 
--- READ ---
c0 01 03 04 00 20 40 16 00 01 05 00 00 c0 DEBUG: Error: 
DEBUG: INVALID_COMMAND
DEBUG: 

--- WRITE ---
c0 00 03 10 04 6c 00 00 00 00 04 00 00 27 05 00 00 00 00 00 00 00 00 00 00 3e 85 63 75 f7 02 b7 86 11 3c 37 86 11 3c 37 85 11 3c 41 11 93 86 06 ff 13 06 86 d2 93 05 40 12 13 05 45 0c 06 c6 ef 50 b0 6c 01 45 82 80 01 11 06 ce 22 cc 26 ca 4a c8 4e c6 0d e1 b7 86 11 3c 37 86 11 3c 93 86 06 56 13 06 06 e5 93 05 a0 36 37 85 11 3c 13 05 45 0c ef 50 90 69 89 47 2e 89 63 fd b7 00 b7 86 11 3c 37 86 11 3c 93 86 06 57 13 06 06 e5 93 05 b0 36 e1 bf aa 84 63 85 f5 00 93 04 35 00 f1 98 93 05 db dc 0e 05 45 ef 50 d0 7e 2a 84 26 85 ef 50 10 79 aa 89 11 db dc 05 e1 22 85 ef 50 d0 78 4e 85 ef 50 70 78 01 44 22 85 f2 40 62 44 d2 44 42 49 b2 49 05 61 82 80 93 06 c4 03 0d 47 01 46 81 45 05 45 ef 00 70 47 93 06 04 09 0d 47 01 46 81 45 05 45 ef 00 70 46 ce 86 22 86 ca 85 26 85 ef f0 1f 8a d1 b7 39 71 06 de 22 dc 26 da 4a d8 4e d6 52 d4 56 d2 5a d0 5e ce 62 cc 66 ca 0d e1 b7 86 11 3c 37 86 11 3c 93 86 86 5b 13 06 46 e6 93 05 30 40 37 85 11 3c 13 05 45 0c ef 50 10 5d 2a 84 2e 8b b2 84 b6 89 89 ed 19 ce b7 86 11 3c 37 86 11 3c 93 86 06 59 13 06 46 e6 93 05 40 40 d1 bf 5c 41 63 ee c7 06 1c 44 89 8b 99 c3 05 49 ad c8 ef 10 90 4d 33 0c 35 01 4e 8a 93 0a c4 03 85 4b fd 5c d2 85 56 85 ef 10 20 38 63 1a 75 05 ef 20 b0 7b 5c 44 a6 85 22 85 82 97 2a 89 63 17 75 03 1c 48 26 86 da 85 22 85 82 97 22 85 ef f0 bf e5 2a c6 ef 20 10 7d 32 45 31 e1 81 46 01 46 81 45 13 05 04 09 ef 00 90 57 21 a8 63 86 99 01 ef 10 b0 47 33 0a ac 40 ef 20 d0 7a e3 f4 49 fb 01 49 f2 50 62 54 4a 85 d2 54 42 59 b2 59 22 5a 92 5a 02 5b f2 4b 62 4c d2 4c 21 61 82 80 81 46 01 46 81 45 56 85 ef 00 90 53 4d bf 79 71 06 d6 22 d4 26 d2 4a d0 4e ce 0d e1 b7 86 11 3c 37 86 11 3c 93 86 86 5b 13 06 46 e7 93 05 00 44 37 85 11 3c 13 05 45 0c ef 50 70 4d 2a 84 ae 8 [... cut ...]
- Add an option to bypass the control of the GPIO's (usefull when you want to do it "by hands" or to test the "flashing" like now)

Here is the port files: [linux_port.zip](https://github.com/espressif/esp-serial-flasher/files/13450953/linux_port.zip)

But during my tests, I was putting the ESP32-C3 in download mode manually by pressing Reset/Boot buttons.

Currently I am trying this solution on a computer to evaluate it, because at terms I will have to integrate a solution to update the firmware of an ESP-C3 chip used for WiFi/BLE in a Yocto project.

DNedic, 2023-11-28:
Hello @caipiblack , upon taking a look at the logs I don't see why the responses would be `INVALID_COMMAND`. It would be useful if you could provide logic analyzer traces or take a look at what is actually happening yourself.

Also, please make sure your power supply for the ESP board is strong and stable enough to not cause spurious resets, as those might cause the output we are seeing in the logs.

Lastly, we are very interested in your modification of the `Raspberry` port, please consider submitting a PR, with it being maintained as an official port you would have less maintainance burden as well.

dobairoland, 2023-12-11:
I'm closing it for inactivity. We'll reopen if any new information is provided.

closed 2023-12-11 by dobairoland