espressif/esp-serial-flasher #89: slave device PINOUT

cyberdevnet, 2023-12-06:
### Port

ESP

### Target chip

ESP32-S3

### Hardware Configuration

ESP32-8048S043

### Log output

```plain
connect to target failed (because no physical connection yet)
```

### More Information

Maybe a stupid question but, I'm trying to flash a display model ESP32-8048S043 (e.g. https://it.aliexpress.com/i/1005004788147691.html) which has no exposed GPIO0 and RESET pin, what can I do?

JimNickerson, 2023-12-06:
Have you plugged the USB cable into your computer ?

cyberdevnet, 2023-12-06:
@JimNickerson yes it connected with the usb cable at the moment, it takes power from usb (but this will change)

JimNickerson, 2023-12-06:
what programming environment are you using that produces the
 "connect to target failed (because no physical connection yet)" message ?

JimNickerson, 2023-12-06:
are you running one of the example ino from the .zip ?

JimNickerson, 2023-12-06:
maybe you could begin with one of the Arduino example ino in the zip available from the ali website.
As this is an S3 and a USB cable ask arduino to connect to the USB comms port and program the board.
Espressif documents the programming sequence when using the S3 USB.

JimNickerson, 2023-12-06:
Are you trying to turn the Ali ESP32-8048S043 board into a Serial Flasher Device ?????

JimNickerson, 2023-12-06:
What do you expect the "Serial Flasher example" to do when you have completed your project ?

DNedic, 2023-12-07:
Hello @cyberdevnet , It would help us provide better feedback for you if you provided your usecase and intended connection setup. `esp-serial-flasher` is a library for flashing Espressif SoC's from other MCUs and SBCs, but if all you want to do is to flash an Espressif SoC offline, you can use `esptool` or `espflash`.

For a quickstart, take a look at [ESP-IDF Get Started](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html) section of our documentation.

If you do indeed have an usecase for `esp-serial-flasher`, please provide more details.

dobairoland, 2023-12-18:
I'm closing this for inactivity.

closed 2023-12-18 by dobairoland

cyberdevnet, 2023-12-18:
Sorry for the late response, let me explain my setup and what I'm trying to achieve.

Setup Description:

Master Device: ESP32-S3-DevKit C N16R8
Slave Device: Display model ESP32-8048S043
Connection: Serial interface (Serial1) between Master and Slave

I have an ESP32-S3-DevKit C N16R8 which is my master device, the master is connected via serial interface (Serial1) to a display model ESP32-8048S043 (which is the slave), with Serial1 I can share data between the Master and the Slave. This Serial connection is used to update the display but also to send data from the display to the master. Note that Master and Slave are connected as follow:

Master    Slave
RX              TX
TX              RX
+5v            +5v
GND          GND

Why I need esp-serial-flasher?
The firmware of the master device is upgraded using either its USB Serial connection or via OTA, which is perfectly fine for my needs. BUT I would like to do also the following:
- Download the display firmware (hosted on a web server) using the master ESP32
- Save the downloaded display firmware on master flash 
- Use esp-serial-flasher to transfer the display firmware from the master to slave
- Upgrade the display

My concern is not about the code but about the physical connection between the master and the slave. As seen in this image [https://macsbug.files.wordpress.com/2022/10/4280s043_layout.jpg?w=700], the slave display lacks exposed RST and IO0 pins.

Should I solder 2 cables at the BOOT and RST button? 

Is there any other way to put the slave into boot mode without using RST and IO0 pin?

I hope my question is now clear, any guidance or suggestions would be greatly appreciated.

DNedic, 2023-12-18:
Hello @cyberdevnet , unfortunately there is no other way to get the device into the Boot ROM without pulling down the strapping pin (IO0). USB can set RESET and BOOT as well by using DTR and RTS, however there is no support at the moment for the USB interface in `esp-serial-flasher` (see: https://github.com/espressif/esp-serial-flasher/issues/64 for the feature request).

cyberdevnet, 2023-12-18:
@DNedic Thanks for your answer, I even tried to put the device into boot mode programmatically bridging another unused gpio for the reset and helding LOW gpio0, the device goes into a reset loop :( 

I really wanted avoid soldering flying wire on that display