espressif/esp-serial-flasher #90: Cannot connect to target. Error: 2 (ESP-12F ESP8266 module)

wosayttn, 2023-12-11:
### Port

Nuvoton MCU

### Target chip

ESP8266

### Hardware Configuration

What dev boards/custom PCB are you using :  NuMaker-IoT-M467
how are the chips connected:  M467   
which baudrate are you trying to flash with?    115200N81

### Log output

```plain
--- WRITE ---
C0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 C0
--- READ ---
C0 08 24 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 C0
--- WRITE ---
C0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 C0
--- READ ---
C0 08 24 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 C0
Cannot connect to target. Error: 2
done
```

### More Information

I see the the command response from ESP8266, and it is not a READ_DIRECTION at first byte.
Does esp-serial-flasher support ESP-12F module?

```c
    do {
        err = SLIP_receive_packet(resp, resp_size);
        if (err != ESP_LOADER_SUCCESS) {
            return err;
        }
    } while ((response->direction != READ_DIRECTION) || (response->command != cmd));
````

DNedic, 2023-12-14:
Hello @wosayttn , after taking a look at the datasheet for the module we are using, I don't see any reason why it would not work with `esp-serial-flasher`, the RX, TX, IO0 and RST pins are all exposed.

From the log, there indeed is a lack of the direction byte, which is something I don't think I've seen so far. Can you obtain the information about the rom version by connecting an USB to serial converter to the module, then resetting it with the IO0 pulled down? That would give us further clues.

Additionally, perhaps you could take a logic analyzer trace to confirm that the byte is indeed missing.

wosayttn, 2023-12-15:
@DNedic yup. You are right. I can't see the READ_DIRECTION byte using LA, but I solved the issue after pull-up UART RX pin. Thanks.

wosayttn, 2023-12-18:
Hi, We still get some trouble on flash_binary function. We want to use the esp-serial-flasher instead of esptool on PC side. below is my implementation, It can flash all firmware files of AT_NONOS_V1.7 successfully, but AT function can't workable after a reboot. I miss something?

[image]

```c
#include "NuMicro.h"
#include "esp_loader.h"
#include "example_common.h"

#define STR2(x) #x
#define STR(x) STR2(x)
#define INCBIN(name, file) \
    __asm__(".section .rodata\n" \
            ".global incbin_" STR(name) "_start\n" \
            ".balign 16\n" \
            "incbin_" STR(name) "_start:\n" \
            ".incbin \"" file "\"\n" \
            \
            ".global incbin_" STR(name) "_end\n" \
            ".balign 1\n" \
            "incbin_" STR(name) "_end:\n" \
            ".byte 0\n" \
    ); \
    extern const __attribute__((aligned(32))) void* incbin_ ## name ## _start; \
    extern const void* incbin_ ## name ## _end; \

#define PATH_BOOT_BIN            "..//ATFW_v17//boot_v1.7.bin"
#define ADDR_BOOT_BIN            0x000000U
INCBIN(boot, PATH_BOOT_BIN);

#define PATH_APP_BIN             "..//ATFW_v17//user1.2048.new.5.bin"
#define ADDR_APP_BIN             0x001000U
INCBIN(app, PATH_APP_BIN);

#define PATH_PART_BIN            "..//ATFW_v17//esp_init_data_default_v08.bin"
#define ADDR_PART_BIN            0x1FC000U
INCBIN(part, PATH_PART_BIN);

#define PATH_BLANK_BIN           "..//ATFW_v17//blank.bin"
#define ADDR_BLANK1_BIN          0x0FE000U
#define ADDR_BLANK2_BIN          0x1FE000U
INCBIN(blank, PATH_BLANK_BIN);

void esp8266_fw_import(void)
{
    partition_attr_t boot;
    partition_attr_t part;
    partition_attr_t app;
    partition_attr_t blank1;
    partition_attr_t blank2;

    boot.data = (const uint8_t *)&incbin_boot_start;
    boot.size = (uint32_t)((char *)&incbin_boot_end - (char *)&incbin_boot_start);
    boot.addr = ADDR_BOOT_BIN;

    app.data  = (const uint8_t *)&incbin_app_start;
    app.size  = (uint32_t)((char *)&incbin_app_end - (char *)&incbin_app_start);
    app.addr  = ADDR_APP_BIN;

    part.data = (const uint8_t *)&incbin_part_start;
    part.size = (uint32_t)((char *)&incbin_part_end - (char *)&incbin_part_start);
    part.addr = ADDR_PART_BIN;

    blank1.data  = (const uint8_t *)&incbin_blank_start;
    blank1.size  = (uint32_t)((char *)&incbin_blank_end - (char *)&incbin_blank_start);
    blank1.addr  = ADDR_BLANK1_BIN;

    blank2.data  = (const uint8_t *)&incbin_blank_start;
    blank2.size  = (uint32_t)((char *)&incbin_blank_end - (char *)&incbin_blank_start);
    blank2.addr  = ADDR_BLANK2_BIN;

    printf("[1] Loader: DataAddr:%p, Size: %d, FlashAddr: %08x\n", boot.data, boot.size, boot.addr);
    flash_binary(boot.data,    boot.size,    boot.addr);

    printf("[2] Application: DataAddr:%p, Size: %d, FlashAddr: %08x\n", app.data, app.size, app.addr);
    flash_binary(app.data,     app.size,     app.addr);

    printf("[3] Partition: DataAddr:%p, Size: %d, FlashAddr: %08x\n", part.data, part.size, part.addr);
    flash_binary(part.data,    part.size,    part.addr);

    printf("[4] Blank1: DataAddr:%p, Size: %d, FlashAddr: %08x\n", blank1.data, blank1.size, blank1.addr);
    flash_binary(blank1.data,  blank1.size,  blank1.addr);

    printf("[5] Blank2: DataAddr:%p, Size: %d, FlashAddr: %08x\n", blank2.data, blank2.size, blank2.addr);
    flash_binary(blank2.data,  blank2.size,  blank2.addr);
}
```
```bash
+------------------------------------------------------------------+
|              ESP8266-12F WiFi firmware flasher                   |
+------------------------------------------------------------------+
Connected to target
Target Chip is ESP8266
[1] Loader: DataAddr:001045c0, Size: 4080, FlashAddr: 00000000
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
[2] Application: DataAddr:001055c0, Size: 455508, FlashAddr: 00001000
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
[3] Partition: DataAddr:00174920, Size: 128, FlashAddr: 001fc000
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
[4] Blank1: DataAddr:001749b0, Size: 4096, FlashAddr: 000fe000
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
[5] Blank2: DataAddr:001749b0, Size: 4096, FlashAddr: 001fe000
Erasing flash (this may take a while)...
Start programming
Progress: 100 %
Finished programming
Done
```

DNedic, 2023-12-18:
Did you verify the device actually did reboot? This could be related to https://github.com/espressif/esp-serial-flasher/issues/76, a known issue where the code doesn't start executing when flashed after `FLASH_END` (this problem does not exist when uploading to RAM). Try doing a hardware reset instead for a start.

wosayttn, 2023-12-20:
@DNedic  Yes, I did. After the flashing is complete and reboot, the ESP8266 becomes a brick. 

```c
void esp8266_flasher(void)
{
    partition_attr_t boot;
    partition_attr_t part;
    partition_attr_t app;
    partition_attr_t blank1;
    partition_attr_t blank2;

    boot.data = (const uint8_t *)&incbin_boot_start;
    boot.size = (uint32_t)((char *)&incbin_boot_end - (char *)&incbin_boot_start);
    boot.addr = ADDR_BOOT_BIN;

    app.data  = (const uint8_t *)&incbin_app_start;
    app.size  = (uint32_t)((char *)&incbin_app_end - (char *)&incbin_app_start);
    app.addr  = ADDR_APP_BIN;

    part.data = (const uint8_t *)&incbin_part_start;
    part.size = (uint32_t)((char *)&incbin_part_end - (char *)&incbin_part_start);
    part.addr = ADDR_PART_BIN;

    blank1.data  = (const uint8_t *)&incbin_blank_start;
    blank1.size  = (uint32_t)((char *)&incbin_blank_end - (char *)&incbin_blank_start);
    blank1.addr  = ADDR_BLANK1_BIN;

    blank2.data  = (const uint8_t *)&incbin_blank_start;
    blank2.size  = (uint32_t)((char *)&incbin_blank_end - (char *)&incbin_blank_start);
    blank2.addr  = ADDR_BLANK2_BIN;

    printf("[1] Loader: DataAddr:%p, Size: %d, FlashAddr: %08x, crc32: %08x\n", boot.data, boot.size, boot.addr, crc32((uint32_t*)boot.data, (uint32_t)boot.size));
    flash_binary(boot.data,    boot.size,    boot.addr);

    printf("[2] Application: DataAddr:%p, Size: %d, FlashAddr: %08x, crc32: %08x\n", app.data, app.size, app.addr, crc32((uint32_t*)app.data, (uint32_t)app.size));
    flash_binary(app.data,     app.size,     app.addr);

    printf("[3] Partition: DataAddr:%p, Size: %d, FlashAddr: %08x, crc32: %08x\n", part.data, part.size, part.addr, crc32((uint32_t*)part.data, (uint32_t)part.size));
    flash_binary(part.data,    part.size,    part.addr);

    printf("[4] Blank1: DataAddr:%p, Size: %d, FlashAddr: %08x, crc32: %08x\n", blank1.data, blank1.size, blank1.addr, crc32((uint32_t*)blank1.data, (uint32_t)blank1.size));
    flash_binary(blank1.data,  blank1.size,  blank1.addr);

    printf("[5] Blank2: DataAddr:%p, Size: %d, FlashAddr: %08x, crc32: %08x\n", blank2.data, blank2.size, blank2.addr, crc32((uint32_t*)blank2.data, (uint32_t)blank2.size));
    flash_binary(blank2.data,  blank2.size,  blank2.addr);

    esp_loader_flash_finish(false);

    esp_loader_reset_target();
}

```

wosayttn, 2023-12-20:
Yes, I use ESP8266 Download tool v3.6.5 to flash these binaries, it is workable. I also check CRC32 checksum of these images in our MCU flash, all are right. below screenshot is the AT v1.7 firmware image layout, all addresses are match. Does esp-serial-flasher read back all content on ESP8266's flash? (or other ESP tool can do it?)

[image]

DNedic, 2023-12-20:
You can read the flash using `esptool`, yes. We currently do not have flash reading implemented in `esp-serial-flasher` (and it would be slower without the stub anyway).

dobairoland, 2024-01-15:
I'm closing for inactivity.

closed 2024-01-15 by dobairoland