espressif/esp-serial-flasher #91: 官方c源码实现烧录功能未能成功

Qfeng116, 2023-12-13:
在使用官方https://github.com/espressif/esp-serial-flasher的C源码时，实现不了烧录功能。
以下为esp32c3模块串口抓到C源码对其发送的数据流：
[image]
[image]
[image]
[image]
上面的图片可以看出pc端的烧录工具和C源码烧录功能串口发出的数据流是有一定差距的。
以下为C源码的使用过程：
-----------------------------------------------------------
传入的filepath为"../binaries/bin_1027/bootloader.bin"
int  espmcu_loader(const char* filepath)
{
    int ret = 0;
    ret = upload_file(filepath, 0x0);
    return ret;
}
----------------------------------------------------------->
static int upload_file(const char *path, size_t address)
{
    char *buffer = NULL;
    FILE *image = fopen(path, "rb");

    if (image == NULL) {

        return -1;
    }

    fseek(image, 0L, SEEK_END);
    size_t size = ftell(image);
    rewind(image);

    buffer = (char *)malloc(size);

    if (buffer == NULL) {
        printf("Error: Failed allocate memory\n");
        goto cleanup;
    }

    // copy file content to buffer
    size_t bytes_read = fread(buffer, 1, size, image);

    if (bytes_read != size) {
        printf("Error occurred while reading file");
        goto cleanup;
    }

    flash_binary((const uint8_t *)buffer, size, address);

cleanup:
    fclose(image);
    free(buffer);

    return 0;
}
----------------------------------------------------------->
#ifdef SERIAL_FLASHER_INTERFACE_UART
esp_loader_error_t flash_binary(const uint8_t *bin, size_t size, size_t address)
{
    esp_loader_error_t err;
    static uint8_t payload[1024];   //2048->1024
    const uint8_t *bin_addr = bin;

    err = esp_loader_flash_start(address, size, sizeof(payload));

    if (err != ESP_LOADER_SUCCESS) {
        printf("Erasing flash failed with error %d.\n", err);
        return err;
    }
        printf("Start programming\n");
    size_t binary_size = size;
[... 346 lines cut ...]
    return ESP_LOADER_SUCCESS;
}
-----------------------------------------------------------
以上代码中的send_cmd：
esp_loader_error_t send_cmd(const void *cmd_data, uint32_t size, uint32_t *reg_value)
{
    response_t response;
    command_t command = ((const command_common_t *)cmd_data)->command;
    RETURN_ON_ERROR( SLIP_send_delimiter() );
    RETURN_ON_ERROR( SLIP_send((const uint8_t *)cmd_data, size));
    RETURN_ON_ERROR( SLIP_send_delimiter() );

    //return check_response(command, reg_value, &response, sizeof(response));

    const uint8_t response_cnt = command == SYNC ? 8 : 1;

        for (uint8_t recv_cnt = 0; recv_cnt < response_cnt; recv_cnt++) {
            RETURN_ON_ERROR(check_response(command, reg_value, &response, sizeof(response)));
        }

        return ESP_LOADER_SUCCESS;
}
===================================================================
附代码：
[Uploading esp32-app.zip…]()

dobairoland, 2023-12-13:
@Qfeng116 Can you please help to translate the issue report to English? I'm sorry for the inconvenience but we cannot put into a translator software the screenshots. Thank you for understanding!

Qfeng116, 2023-12-13:
@dobairoland ok,my friend, thank you for your reply. The following is the translation corresponding to the screenshot, and the blue small character near the Chinese character on the screenshot is the corresponding translation.1.
C源代码发送串行数据流C source code sent serial data flow
PC刻录工具发送串行数据流PC burning toolSend serial data flow
C源代码没发，PC有发C source code did not send, PC has sent
源代码后发送的刻录命令命令是错误的C source code sent the burning command after the command is wrong
这部分命令试图转换代码中的数据以模仿pc发送的数据，但不确定这种转换数据的方式本身是否不适用于其他芯片或bin升级固件This section of the command attempts to convert data in the code to imitate the data sent by the pc, but it is not sure whether this way of converting data by itself will not be applicable to another chip or bin upgrade firmware
2.
C源码发的串口数据流C source code sent serial data flow
PC烧录工具发的串口数据流PC burning tool sent serial data flow
发命令读写完寄存器后，PC烧录工具还发了一段READ REG命令，不确定其作用After the command is finished reading and writing the register, the PC burning tool also sends a READ REG command, unsure of its role
3.
C源码发的串口数据流C source code sent serial data flow
PC烧录工具发的串口数据流PC burning tool sent serial data flow
C源码后续没有发与PC端相类似的命令了C source code did not send a similar command with the PC terminal
发完上图提到的一段后，发的命令应该是解析bin文件包After sending the paragraph mentioned above, the command sent should be parsing binDocument case
发SPI_FLASH_MD5命令Send the SPI_FLASH_MD5 command
继续解析Continue to analyze
4.
C源码发的串口数据流C source code sent serial data flow
PC烧录工具发的串口数据流PC burning tool sent serial data flow
最后发烧录命令Finally send the burn command
[Translated graph.zip](https://github.com/espressif/esp-serial-flasher/files/13660350/Translated.graph.zip)

DNedic, 2023-12-14:
Hello @Qfeng116 , after taking a look at the issue it is not clear to me what problem you are facing and what you're comparing the data stream with. I presume you are comparing it to `esptool` flashing, and in that case keep in mind `esptool` at the moment works quite differently, namely:

1. It uses the `GET_SECURITY_INFO` command whereas we do not yet
2. It uploads the flasher stub to RAM which has it's own commands not supported in the Boot ROM we are talking to directly
3. The flasher stub has it's own commands which the boot rom does not support
4. `esptool` handles SPI attach and parameter setting differently
5. Probably more things I didn't think of

You can run `esptool` with the `--no-stub` argument to get a more similar flashing procedure that only talks to the Boot ROM, but regardless, it would be useful to get both RX and TX logs of the failing flashing process, so we can get to the bottom of the issue.

Qfeng116, 2023-12-14:
Hello @DNedic ,thank u for the response.Let me add a further description of the problem I face.The data stream of the pc burning tool is to burn the esp32c3 module using the PC burning tool flash_download_tool_3.9.5. After connecting the burning line to be connected, pull out two lines to observe and record the data stream received by the rx and tx serial ports. C source code burning function after the release of the data flow is also obtained through the two lines of rx, tx observation.
[image]
[image]
[image]
[image]

[image]

Qfeng116, 2023-12-14:
Hello@DNedic, so this C source code(https://github.com/espressif/esp-serial-flasher) can achieve burning function?

Qfeng116, 2023-12-14:
Hello@DNedic ,Can you provide the c source code that can realize the burn function, so that I can directly merge the burn function? I want to explore the efficiency of implementing this function is too low, and the uncertainty is too high, which is not conducive to the later maintenance of the function.

DNedic, 2023-12-14:
I see that the response to `FLASH_START` is an unknown error, `failed` is set to 1 but the error code is not set. Perhaps you could provide the binary so we can try to reproduce the issue? 

Addiitonally, you could try disconnecting everything except the connection to the PC you are using to flash with `esp-serial-flasher`, just in case. And if you have a logic analyzer trace, it might help us as well.

Qfeng116, 2023-12-14:
[bootloader.zip](https://github.com/espressif/esp-serial-flasher/files/13672070/bootloader.zip)
Ok, the above is the burned bin file, the pc side of the esp-serial-flasher can be burned normally regardless of whether there is an additional set of rx, tx wiring.

DNedic, 2023-12-14:
@Qfeng116 I have successfully flashed the file you attached with the ESP port, there seems to be nothing wrong with it, it looks like the issue is most likely in your port (Qt on Windows I assume from your provided screenshots).

If you can, please provide a logic analyzer trace of the data transferred, it could give us more clues.

Qfeng116, 2023-12-15:
[Logic analyzer record.zip](https://github.com/espressif/esp-serial-flasher/files/13680465/Logic.analyzer.record.zip)
@DNedic Above is my code click burn function after the logic analyzer to grab the transmission data record.The port I used was preset. I also tried to copy the serial port data stream directly and send it to the module for simulation and verification at the same time, and compared whether the returned serial port data stream was consistent with strcmp before sending the next data stream. Through the above method, I successfully simulated the erasure function, because the serial port data stream received and received by the erasure function was roughly unchanged. And there's not a lot of data flow. However, if the same way on the burning function will not work, the burning function of the data stream is huge, and change the bin file to burn the data stream will also change.

DNedic, 2023-12-15:
@Qfeng116 I can't really conclude anything from this logic analyzer trace, there is no response after `FLASH_BEGIN` and the previous commands did not error out.

Qfeng116, 2023-12-16:
[image]
@DNedic Yes, this is strange, and after sending FLASH_BEGIN, the module returns a data stream with error codes.

Qfeng116, 2023-12-18:
Hello@DNedic ,May I ask if there is something wrong with the process of my C source code, I use the example of Raspberry PI in the following link[https://github.com/espressif/esp-serial-flasher](https://github.com/espressif/esp-serial-flasher%E7%9A%84C%E6%BA%90%E7%A0%81%E6%97%B6%EF%BC%8C%E5%AE%9E%E7%8E%B0%E4%B8%8D%E4%BA%86%E7%83%A7%E5%BD%95%E5%8A%9F%E8%83%BD%E3%80%82)

DNedic, 2023-12-18:
There should be nothing wrong with the raspberry port, what hardware setup are you using?

Qfeng116, 2023-12-18:
I use linux system, mcu is ssd212.

Qfeng116, 2023-12-19:
Hello@DNedic ,maybe you are not very clear about my problem now, is this, I want to do a burn function: you can upgrade the esp32c3 through the MCU, just like on Windows QT screenshot click burn button, will start to perform the burn function (module will enter burn mode in advance). However, I failed to implement the functionality with the sample code and wondered if there was something wrong with my program flow. At present, I want to implement the function in the QT simulation on Windows first, if it is realized, the function will be moved to the overall code call, and after triggering the burn buttom, the software will pull the module to enter the upgrade mode required pins, and then execute the burn function.

Qfeng116, 2023-12-20:
Hello@DNedic, perhaps I have not described enough, if you need to provide more information, please let me know, I will do my best to provide.

dobairoland, 2023-12-21:
@Qfeng116 We struggle to understand exactly what is your problem and what are you trying to accomplish.

Lets clarify a couple of things:
1. What kind of burn function do you want to implement? This might be just a misunderstanding in terminology but we usually refer to burning of eFuses. Please correct me if I'm wrong but you want to flash regular flash chip. Nothing more fancy.
2. Please don't try to reproduce by your code the communication of the Serial Flasher Tool. As it was pointed out by @DNedic, the communication is quite different in comparison with esp-serial-flasher. If you take the https://github.com/espressif/esp-serial-flasher/tree/master/examples/raspberry_example project then it will use the esp-serial-flasher library which implements all the necessary communication. You don't have to implement it by yourself.
3. If you try to do something which is not supported by esp-serial-flasher then please describe it clearly, i.e. what should that "burning function" do?
4. If esp-serial-flasher is not behaving as it should then please describe the situation clearly. We need in this case what you are trying to accomplish and how does esp-serial-flasher behaving instead. We will need in this case the complete communication trace of the esp-serial-flasher project (not that of the Serial Flasher Tool).

Qfeng116, 2023-12-22:
@dobairoland The feature I want to implement is a c code interface to upgrade the new firmware for the esp32c3. I can give this code the address of the bin file I want to upgrade, and this code burns the bin file into the esp32c3.

dobairoland, 2023-12-22:
> The feature I want to implement is a c code interface to upgrade the new firmware for the esp32c3. I can give this code the address of the bin file I want to upgrade, and this code burns the bin file into the esp32c3.

You don't have to implement that by yourself. Everything is shown in this example how to do it: https://github.com/espressif/esp-serial-flasher/tree/master/examples/raspberry_example

Moreover, @DNedic have confirmed earlier that he could flash the BIN file you provided earlier.

So where is the issue? Does the RPI example liked above updated with your binary fail at any steps? If yes then please provide the full console output and the trace capturing this communication.

Qfeng116, 2023-12-22:
@dobairoland I used this example, but since I was using qt read and write instead of Raspberry PI read and write, I also preset the baud rate and port, as shown in the image below, click Connect and then click Close to enter the Erase Burn button interface.
[image]
There is no loader_port_enter_bootloader() on the overall code process because my external hardware is already in burn mode.
The serial read and write functions SendData(), ReadData() and wait_read () are defined below:
void SerialTool::SendData(const char* data,int length)
{
    QByteArray send_data;
    for(int i=0; i<length; i++)
    {
       send_data.append(data[i]);
    }

    qDebug() << "Send Data:" << send_data.toHex() << "-Length:" << send_data.size();
    emit serial_send(send_data);
}
//=========================line between==========================
int SerialTool::ReadData(char* data,qint64 maxlen)
{
    return m_serial->ReadPort(data,maxlen);
}
int SerialPortQt::ReadPort(char* data,qint64 maxlen)
{
    int length = -1;

    if(RecvCall != NULL)
    {
       return -1;
    }

    length = recdata.size();

    if((length < maxlen)||(length <=0 ))
    {
        endflag=1;
        return -1;
    }

    for(int i = 0; i<maxlen; i++)
    {
       data[i] = recdata[i];
       cnt++;
    }
    cnt=0;
    length = maxlen;
    recdata.remove(0,length);

    return length;
}
//=========================line between==========================
int SerialTool::wait_read(int delay)
{
    m_serial->waitms(delay);
    return 0;
}
void SerialPortQt::waitms(int delay)
{
    port->waitForReadyRead(delay);
}
bool waitForReadyRead(int msecs = 30000) override;
//=========================line between==========================

In short, the code is similar to the example code, but I can only compare the errors in the code steps through serial data stream. As can be seen from the following figure, after my code completes loader_flash_begin_cmd(), check_response() returns an error code, and it is an unknown error. So I'm not sure what went wrong with the steps, because the steps are the same as in the sample code.
[image]

dobairoland, 2023-12-22:
Ok, I think I understand the situation now. The confusion was caused because we assumed that something is wrong with esp-serial-flasher. But instead you have trouble with your own Qt-based application.

I'm regret to tell you that we cannot help with that. We have limited resources and we have to use them to fix issues and implement new features in esp-serial-flasher. As I stated before, if you demonstrate an issue with it then we are happy to return to. But until then I'm closing this issue. I'm sorry.

closed 2023-12-22 by dobairoland

Qfeng116, 2023-12-22:
@dobairoland But my qt send and receive data is no problem, otherwise it can not use the direct data stream to achieve the erasure function.

Qfeng116, 2023-12-25:
@dobairoland I want to know whether my rewrite the https://github.com/espressif/esp-serial-flasher code is wrong, because I there is no question of qt on serial port to send and receive data.

Qfeng116, 2024-01-18:
@dobairoland Maybe you didn't get the message,my qt send and receive data is no problem, otherwise it can not use the direct data stream to achieve the erasure function.I want to know whether my rewrite the https://github.com/espressif/esp-serial-flasher code is wrong, because I there is no question of qt on serial port to send and receive data.

zrb2002, 2024-01-23:
@dobairoland Can we look at that again

zrb2002, 2024-01-23:
@dobairoland hi，Excuse me, how is the problem going now

dobairoland, 2024-01-23:
I've left a comment earlier - https://github.com/espressif/esp-serial-flasher/issues/91#issuecomment-1867460452. We are not working on this anymore.