espressif/esp-serial-flasher #99: Cannot connect to target. Error: 2 - ESP32 Example, Wired up correctly.

GreenDiscord, 2024-03-06:
### Port

ESP

### Target chip

ESP32-WRROM-32U

### Hardware Configuration

I'm using a DOIT Esp32 DevKit v1, Connected through Dupont/breadboard cables. Baudrate I believe is 230400.
I'm trying to flash my own bin files which are known to work with an ESP-PROG.

### Log output

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q�ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15532
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (29) boot: ESP-IDF v5.2 2nd stage bootloader
I (29) boot: compile time Mar  6 2024 18:21:01
I (29) boot: Multicore bootloader
I (33) boot: chip revision: v3.1
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000       
I (82) boot:  2 factory          factory app      00 00 00010000 00200000       
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1c9054h (1871956) map
I (744) esp_image: segment 1: paddr=001d907c vaddr=3ffb0000 size=02254h (  8788) load
I (747) esp_image: segment 2: paddr=001db2d8 vaddr=40080000 size=04d40h ( 19776) load
I (757) esp_image: segment 3: paddr=001e0020 vaddr=400d0020 size=189a0h (100768) map
I (793) esp_image: segment 4: paddr=001f89c8 vaddr=40084d40 size=08a20h ( 35360) load
I (813) boot: Loaded app from partition at offset 0x10000
I (814) boot: Disabling RNG early entropy source...
I (825) cpu_start: Multicore app
I (834) cpu_start: Pro cpu start user code
I (834) cpu_start: cpu freq: 160000000 Hz
I (834) cpu_start: Application information:
I (837) cpu_start: Project name:     esp-serial-flasher
I (843) cpu_start: App version:      v1.2.0-1-gc1c0878-dirty
I (849) cpu_start: Compile time:     Mar  6 2024 18:20:21
I (855) cpu_start: ELF file SHA256:  16af93f9d...
I (861) cpu_start: ESP-IDF:          v5.2
I (865) cpu_start: Min chip rev:     v0.0
I (870) cpu_start: Max chip rev:     v3.99
I (875) cpu_start: Chip rev:         v3.1
I (880) heap_init: Initializing. RAM available for dynamic allocation:
I (887) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (893) heap_init: At 3FFB2FA0 len 0002D060 (180 KiB): DRAM
I (899) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (905) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (912) heap_init: At 4008D760 len 000128A0 (74 KiB): IRAM
I (919) spi_flash: detected chip: generic
I (923) spi_flash: flash io: dio
I (927) main_task: Started on CPU0
I (937) main_task: Calling app_main()
I (937) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (937) gpio: GPIO[32]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---
Cannot connect to target. Error: 2
I (2897) main_task: Returned from app_main()
```

### More Information

If its relevant, im powering the board to be programmed of the 3.3v header on the host board, not 5v. They are built in this way so its known behaviour.

GreenDiscord, 2024-03-06:
[image]
Heres where the board connects.

GreenDiscord, 2024-03-06:
I can tell its not connecting, but I don't get why.

dobairoland, 2024-03-07:
It looks you don't receive anything from the target. Why don't you test the serial connection? You can make a simple application which establishes an UART connection and prints everything it receives. You can toggle the RST pin and see if you receive anything. Until you cannot make this work there is no point trying to flash. You can find UART and GPIO examples as well in the ESP-IDF repo.

closed 2024-03-25 by dobairoland