espressif/esp-serial-flasher #104: Compilation settings for ESP8266 in Arduino IDE

Crystallonus, 2024-03-31:
I'm trying to upload a firmware bin file from an ESP32 to an ESP8266. The test firmware hello world, the bin file of which comes with your library, is uploaded and works, but after I upload the firmware bin file that I make for the ESP8266 in the Arduino IDE, the microcontroller continuously reboots with errors. As I understand, I am using some unsuitable compilation settings, could you suggest what settings should be used? I think it will be useful not only to me. Thank you!

DNedic, 2024-04-02:
Hello @Crystallonus , please enable tracing with SERIAL_FLASHER_DEBUG_TRACE and provide the full log, and if possible, the binary you are trying to flash.

Crystallonus, 2024-04-02:
[DEBUG_TRACE.txt](https://github.com/espressif/esp-serial-flasher/files/14843025/DEBUG_TRACE.txt)
[hello-world.bin.zip](https://github.com/espressif/esp-serial-flasher/files/14843036/hello-world.bin.zip)

My script is just a blink with output to serial, I didn't even try to upload anything more complicated, I started with this and it already didn't work.

`
#include <HardwareSerial.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Started!");
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  Serial.println("loop");
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}

`

DNedic, 2024-04-02:
Considering flashing verification using MD5 is not available for the `ESP8266`, could you try flashing with a very low baudrate (e.g. 9600) just in case and see if the same thing happens?

Crystallonus, 2024-04-03:
> After a brief glance, there seems to be nothing wrong with the binary itself. Considering flashing verification using MD5 is not available for the `ESP8266`, could you try flashing with a very low baudrate (e.g. 9600) just in case and see if the same thing happens?

At a speed of 19200, the result is the same, at a speed of 9600, the upload stops working:
[DEBUG_TRACE2.txt](https://github.com/espressif/esp-serial-flasher/files/14853659/DEBUG_TRACE2.txt)

I tried uploading the blink example compiled under 'non-os-sdk' from platformio, it similarly doesn't work, but now instead of endless rebooting, the board just outputs this log and then nothing happens:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 25408, room 16 
tail 0
chksum 0xd9
load 0x3ffe8000, len 892, room 8 
tail 4
chksum 0xca
load 0x3ffe8380, len 952, room 4 
tail 4
chksum 0x16
csum 0x16
csum err
ets_main.c 
```
When uploading the sketch to the board directly from platformio, it works.

Attaching the debug trace obtained during the upload of this bin and the bin file itself. Speed 19200. Speed 9600 with this bin file also does not work, just like with the previous one.
[hello-world.bin.zip](https://github.com/espressif/esp-serial-flasher/files/14853731/hello-world.bin.zip)
[DEBUG_TRACE3.txt](https://github.com/espressif/esp-serial-flasher/files/14853799/DEBUG_TRACE3.txt)

Just to clarify, I understand correctly that there is no need for me to do anything with the bootloader.bin and partition-table.bin files? I just replace hello-world.bin with mine, right?

DNedic, 2024-04-03:
Yes, if you're actually uploading the bootloader and partition table you built with your app. If you only placed the app binary in the example folder and used preexisting binaries for the bootloader and partition table, unless the partition table matches the layout of your app, it's not gonna work.

Here is the output of `esptool` for the `hello-world.bin` in the example:
```
esptool.py image_info hello-world.bin 
esptool.py vv4.8.dev1
File size: 195984 (bytes)
Detected image type: ESP8266
WARNING: Suspicious segment 0x40210010, length 149844
WARNING: Suspicious segment 0x40234964, length 20164
Image version: 1
Entry point: 40210f08
5 segments

Segment 1: len 0x24954 load 0x40210010 file_offs 0x00000008 [IROM]
Segment 2: len 0x04ec4 load 0x40234964 file_offs 0x00024964 [IROM]
Segment 3: len 0x00620 load 0x3ffe8000 file_offs 0x00029830 [DRAM]
Segment 4: len 0x00a50 load 0x40100000 file_offs 0x00029e58 [IRAM]
Segment 5: len 0x054b0 load 0x40100a50 file_offs 0x0002a8b0 [IRAM]
Checksum: 90 (valid)
```

And here is the output for your binary:
```
esptool.py image_info hello-world.bin
esptool.py vv4.8.dev1
File size: 270464 (bytes)
Detected image type: ESP8266
Image version: 1
Entry point: 4010f480
2 segments

Segment 1: len 0x00d60 load 0x4010f000 file_offs 0x00000008 []
Segment 2: len 0x00028 load 0x3fff20b8 file_offs 0x00000d70 [DRAM]
```

DNedic, 2024-07-02:
Closing this for inactivity.

closed 2024-07-02 by DNedic