espressif/esp-serial-flasher #106: UART does not write data into esp

Sazzad07, 2024-04-24:
### Port

STM32

### Target chip

ESP32C3

### Hardware Configuration

GPI0_08 =1
GPI0_09 = 0
GPI0_02=0
baudate=115200,
no parity and 1 stopbit
[image]

I did not get any any error after Mem data write. However no data is written into the ESP ROM

### Log output

```plain
Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 1
        Listen Port Number         : 61234
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled
        InitWhile                  : Enabled

Waiting for debugger connection...
Debugger connected
Waiting for debugger connection...
Debugger connected
Waiting for debugger connection...
      -------------------------------------------------------------------
                       STM32CubeProgrammer v2.13.0                  
      -------------------------------------------------------------------

Log output file:   C:\Users\SAZZAD~1.SIE\AppData\Local\Temp\STM32CubeProgrammer_a26044.log
ST-Link Server is running on port : 7184
ST-LINK SN  : 003400074D46500720373033
ST-LINK FW  : V3J11M3
Board       : DISCO-H747XI
Voltage     : 3.29V
SWD freq    : 8000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x450
Revision ID : Rev V
Device name : STM32H7xx
Flash size  : 2 MBytes
Device type : MCU
Device CPU  : Cortex-M7/M4
BL Version  : 0x90

Memory Programming ...
Opening and parsing file: ST-LINK_GDB_server_a26044.srec
  File          : ST-LINK_GDB_server_a26044.srec
  Size          : 40.17 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sector 0
Download in Progress:
```

### More Information

I Follow the Flow

Sync-> MEM_Begin -> MEM_DATA -> MEM_END

But don't know why data is not flashed into theESP ROM

  uint32_t start_address = 0x00000000; //ESP
  uint32_t rawDataSize=1024;
  
      static uint8_t mem_begin_cmd[24]=
     { 0x00,0x05,0x10,0x00,0x00,0x00,0x00,0x00,
      0x01, 0x04, 0x00,0x00, //erase size //  0x0401
      0x01, 0x00, 0x00,0x00, // packet count 20U
      0x01,0x04,0x00,0x00,   // packet_size // 0x0401
      0x00,0x00,0x00,0x00,
    };

DNedic, 2024-04-29:
Hello @Sazzad07 , thank you for your submission. Can you use the `SERIAL_FLASHER_DEBUG_TRACE` option and provide the full log? That would give us more insight into what is actually happening.

DNedic, 2024-05-20:
I am closing this for inactivity.

closed 2024-05-20 by DNedic