espressif/esp-serial-flasher #107: Addition of support for firmware flashing directly in binary format

arthurrGama, 2024-04-29:
### Is your feature request related to a problem?

Currently, I'm using the esp-flasher code to flash firmware onto an ESP32. However, I've been facing difficulties when trying to flash another ESP32 with large binaries, so I started using a micro SD card to store the firmware and flash it in parts. The issue is that the code performs flashing in hexadecimal (apparently), and there seem to be restrictions and checks that prevent direct flashing in binary format.

### Describe the solution you'd like

I would like to request the addition of support for direct firmware flashing in binary format. This would make the process more efficient, especially when dealing with large binaries. The ability to read and write firmware directly in binary, without conversion to hexadecimal, would simplify and speed up the flashing process significantly, especially when using a micro SD card.

### Describe alternatives you've considered

Currently, there is no efficient solution for directly flashing large binaries, as the existing code performs flashing in hexadecimal format. We considered using other flashing methods, but none of them seem as practical as modifying the code to support direct flashing in binary.

### Additional context

The project I'm working on involves flashing firmware onto ESP32 using the ESP Flasher code. Adding support for direct flashing in binary would be greatly helpful in simplifying and speeding up the flashing process, especially in cases of large binaries. We appreciate any help or guidance you can provide to implement this functionality.

DNedic, 2024-04-29:
Hello @arthurrGama , we are actually flashing firmware in binary format, it is just being represented as hexadecimal when using the tracing option for easier inspection by humans.

arthurrGama, 2024-04-29:
Hi @DNedic, I think there was a mistake on my part in interpreting the code, sorry for the confusion
I think now I understand the problem I'm having.

arthurrGama, 2024-04-29:
Thanks for your attention!

closed 2024-04-29 by DNedic