espressif/esp-serial-flasher #110: Support Secure Download Mode

matt001k, 2024-05-09:
### Is your feature request related to a problem?

Currently this library does not support secure download mode.
The [esptool](https://github.com/espressif/esptool/blob/master/esptool/cmds.py) library does and detects this properly, then flashes the controller properly.
This tool should support it as well to ensure that secure boot chipsets can be used with this library.

### Describe the solution you'd like

Support secure download mode.

### Additional context

If needed I would be willing to do the work, as this is something that needs to be done ASAP for our project, and it is important to have this feature.
Let me know!

DNedic, 2024-05-09:
Hello @matt001k , thank you for your feature request, I will look into what is needed to provide support for Secure Download Mode.

We are always accepting contributions so if you do create a PR we will work with you to get it to a state where it can be merged and test it.

matt001k, 2024-05-09:
@DNedic thanks for the prompt response.
I would be more than willing to.
Is there any documentation that I could reference to follow the Secure Download loading?
Or would it be mainly be trying to emulate what the esptool.py script is doing?

DNedic, 2024-05-13:
You can read about this feature in the context of flash encryption [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/kconfig.html#config-secure-uart-rom-dl-mode) and the relevant Kconfig option [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/kconfig.html#config-secure-uart-rom-dl-mode), for implementation details you can refer to `esptool` source code.

closed 2024-09-25 by espressif-bot