espressif/esp-serial-flasher #112: Update to 1.3.1 breaks my build

pptz, 2024-06-27:
### Port

esp32

### Target chip

esp32s3

### Hardware Configuration

The problem is with the build, so hardware is not relevant.

### Log output

```plain
/Users/dev/dev/nami-esp/nami-esp/components/nami/src/rcp_upgrade.c: In function 'flash_binary':
/Users/dev/dev/nami-esp/nami-esp/components/nami/src/rcp_upgrade.c:102:11: error: implicit declaration of function 'esp_loader_flash_verify'; did you mean 'esp_loader_flash_write'? [-Werror=implicit-function-declaration]
  102 |     err = esp_loader_flash_verify();
      |           ^~~~~~~~~~~~~~~~~~~~~~~
      |           esp_loader_flash_write
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
```

### More Information

The call to esp_loader_flash_verify() fails to link with 1.3.1. It works with all previous versions.

pptz, 2024-06-27:
[image]
This undefined MD5 and hence the mentioned function. Is this intended or accidental?

DNedic, 2024-07-01:
Hello @pptz , thank you for the submission, It looks like 4ad452dd7e884c233b63d42999964c9af73c788c is the culprit due to the way that MD5_ENABLED is the only option we have not prefixed with `SERIAL_FLASHER_`, and the new `add_option()` function not accounting for that. We will fix this shortly.

wnylei, 2024-07-02:
遇到了同样的问题，  由于 **MD5_ENABLED** 没有定义，导致报错 **esp_loader_flash_verify** 没有声明。

closed 2024-07-02 by espressif-bot