espressif/esp-serial-flasher #114: Cannot connect to target. Error: 2

prabhatel131, 2024-08-29:
### Port

esp32 

### Target chip

esp32s3

### Hardware Configuration

connceted  void slave_monitor(void *arg)
{
#if (HIGHER_BAUDRATE != 115200)
    uart_flush_input(UART_NUM_1);
    uart_flush(UART_NUM_1);
    uart_set_baudrate(UART_NUM_1, 115200);
#endif
    while (1) {
        int rxBytes = uart_read_bytes(UART_NUM_1, buf, BUF_LEN, 100 / portTICK_PERIOD_MS);
        buf[rxBytes] = '\0';
        printf("%s", buf);
    }
}

void app_main(void)
{
    example_binaries_t bin;

    const loader_esp32_config_t config = {
        .baud_rate = 115200,
        .uart_port = UART_NUM_1,
        .uart_rx_pin = GPIO_NUM_16,
        .uart_tx_pin = GPIO_NUM_17,
        .reset_trigger_pin = GPIO_NUM_25,
        .gpio0_trigger_pin = GPIO_NUM_26,
    };
this is configguration

host                   target
16      ---->        tx
17       ->>>       rx
25      ->>>       rst
26     ->>>>      boot

getting error unable to conect target 
what i am dooing wrong

### Log output

```plain
I (0) cpu_start: App cpu up.
I (579) cpu_start: Pro cpu start user code
I (579) cpu_start: cpu freq: 160000000 Hz
I (579) cpu_start: Application information:
I (584) cpu_start: Project name:     esp-serial-flasher
I (590) cpu_start: App version:      v1.5.0-5-g2ac79d7-dirty
I (596) cpu_start: Compile time:     Aug 27 2024 12:09:46
I (602) cpu_start: ELF file SHA256:  10af60cead52ef9d...
I (608) cpu_start: ESP-IDF:          v5.0.5-166-g209a0361c0-dirty
I (615) cpu_start: Min chip rev:     v0.0
I (620) cpu_start: Max chip rev:     v3.99 
I (625) cpu_start: Chip rev:         v3.1
I (630) heap_init: Initializing. RAM available for dynamic allocation:
I (637) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (643) heap_init: At 3FFB2CD8 len 0002D328 (180 KiB): DRAM
I (649) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (655) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (662) heap_init: At 4008CBC4 len 0001343C (77 KiB): IRAM
I (669) spi_flash: detected chip: generic
I (673) spi_flash: flash io: dio
W (677) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (691) app_start: Starting scheduler on CPU0
I (695) app_start: Starting scheduler on CPU1
mI (695) main_task: Started on CPU0
I (705) main_task: Calling app_main()
I (705) gpio: GPIO[25]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (715) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
Cannot connect to target. Error: 2
```

prabhatel131, 2024-08-29:
@jimparis @Chleba @kumekay @pedrominatel

Dzarda7, 2024-08-29:
Hi @prabhatel131, I assume you use ESP32 as a host and ESP32S3 as a target. Can you please try it with different host pins other than GPIO16 and GPIO17 as these might be used by FLASH memory on your board? You can use default pins used in the [example](https://github.com/espressif/esp-serial-flasher/tree/master/examples/esp32_example) to be sure.

dobairoland, 2024-09-09:
I'm closing this for inactivity.

closed 2024-09-09 by dobairoland

prabhatel131, 2024-09-09:
hii @dobairoland @Dzarda7  how you building you 2nd esp src 
as i am not able to build my code in your hello world code  as it not generating any binayry in bineries folder ca you please tell

Dzarda7, 2024-09-20:
Hi @prabhatel131, you can build the binaries using esp-idf. But the binaries are automatically built and bundled with binary for host in case of the examples.

prabhatel131, 2024-09-20:
i want to know how you building your binary automatically ...currently i have to put my binary in that folder after buildung it

Prabhat Singh | Jr. Embedded System Engineer

Dzarda7, 2024-09-20:
There is a script in the example's CMakeLists.txt that runs building and adds the binaries. For example [there](https://github.com/espressif/esp-serial-flasher/blob/master/examples/esp32_example/main/CMakeLists.txt).