espressif/esp-serial-flasher #116: esp32_usb_cdc_acm_example using source file from uSD

versamodule, 2024-09-05:
### Is your feature request related to a problem?

No problem. 

### Describe the solution you'd like

The esp32_usb_cdc_acm_example works fine. But It would be great if there was another example like this that the master device would read the .bin files from a uSD card to flash the slave device.

Dzarda7, 2024-09-05:
Hi @versamodule, thanks for the suggestion, I am planning to the exactly this. I will try to add it in the future.

DNedic, 2024-09-05:
In the meantime, you could check out the [ESP-IDF SD Card Examples](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card) for inspiration.