espressif/esp-serial-flasher #119: Add configurable signal inverts

shiona, 2024-11-05:
### Is your feature request related to a problem?

We are developing a product in which STM32 is flashing an ESP32. Unfortunately there is an inverting connection between the STM32 signal output and the ESP32 enable input.

### Describe the solution you'd like

Probably easiest to have `INVERT_ENABLE`, `INVERT_IO0` or similar defines.
A bit more complex fix, maybe cleaner, would be to have similar flags in the `loader_*_config_t` structs.

I assume nobody would invert the data lines, so those probably don't need the added logic.

### Describe alternatives you've considered

We will make a private fork and make a hack there.

shiona, 2024-11-05:
If either of the solutions are acceptable, I'm more than willing to make a pull request.

Dzarda7, 2024-11-05:
Hi @shiona, thanks for the proposal. I think it might make sense to implement something like this as esptool supports custom reset sequence and esp-serial-flasher is somehow in the middle of the road with supporting configurable BOOT and RESET hold times. This seems reasonable to me. Not sure if keep the established notations and add it as Kconfig option next to the `SERIAL_FLASHER_RESET_HOLD_TIME_MS` and `SERIAL_FLASHER_BOOT_HOLD_TIME_MS` or as another parameter to the structure. @DNedic WHYT?

DNedic, 2024-11-05:
I think it makes sense to have this as a config toggle more, I will explore the idea.

Dzarda7, 2024-11-06:
Hi @shiona you can create PR if you want, seems like quick task. I would add the option [here in Kconfig](https://github.com/espressif/esp-serial-flasher/blob/master/Kconfig) and then put it in each port with inverted logic (in case of stm32 port [here](https://github.com/espressif/esp-serial-flasher/blob/8ece286a3ba09856d56229def5783187594a8aac/port/stm32_port.c#L90)). Thanks.

shiona, 2024-11-06:
Is this https://github.com/shiona/esp-serial-flasher/commit/375eeb211ea886989290c3c6d8c501b9016102b9 a reasonable way to do this? Another possibility would be to define something like `RESET_PIN_{OFF,ON}` and `BOOT_PIN_{BOOTLOADER,APP}` in each port header according to the Kconfig option. That way there wouldn't be those ugly ifdefs in the .c files.

If I knew how to name those values (if reset pin state is `RESET_PIN_OFF`, then is the target device supposed to be running or shut down?) I would have gone that way already, but since I do not, I'm here asking for more help.

Dzarda7, 2024-11-07:
I am against making another define as this will decrease readability. I would either keep the `#ifdef` or do something like `gpio_set_level(s_gpio0_trigger_pin, SERIAL_FLASHER_RESET_INVERT ? 1 : 0);`  and `gpio_set_level(s_gpio0_trigger_pin, SERIAL_FLASHER_RESET_INVERT ? 0 : 1);`.
I personally like the second option I proposed, but I would like to know the opinion of @DNedic so we avoid bothering you with later changes, thanks.

DNedic, 2024-11-07:
> I am against making another define as this will decrease readability. I would either keep the `#ifdef` or do something like `gpio_set_level(s_gpio0_trigger_pin, SERIAL_FLASHER_RESET_INVERT ? 1 : [...]

This is the most reasonable option in my opinion, we want to avoid `#ifdef`s as much as possible and maintain easy to follow code.

closed 2024-11-21 by espressif-bot