espressif/esp-serial-flasher #123: Zephyr example does not build successfully

beriberikix, 2024-11-22:
### Port

Zephyr

### Target chip

ESP32

### Hardware Configuration

esp32_devkitc_wroom

### Log output

```plain
root@52b956611206:/workdir# west build -p -b esp32_devkitc_wroom esp-serial-flasher/examples/zephyr_example/
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /workdir/esp-serial-flasher/examples/zephyr_example
-- CMake version: 3.25.1
-- Found Python3: /opt/venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /root/.cache/zephyr
-- Zephyr version: 4.0.99 (/workdir/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
CMake Warning at /workdir/zephyr/cmake/modules/boards.cmake:110 (message):
  Deprecated BOARD=esp32_devkitc_wroom specified, board automatically changed
  to: esp32_devkitc_wroom/esp32/procpu.
Call Stack (most recent call first):
  /workdir/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /workdir/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /workdir/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Board: esp32_devkitc_wroom, qualifiers: esp32/procpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/root/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/root/zephyr-sdk-0.17.0)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6") 
-- Found BOARD.dts: /workdir/zephyr/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu.dts
-- Found devicetree overlay: /workdir/esp-serial-flasher/examples/zephyr_example/socs/esp32_procpu.overlay
-- Generated zephyr.dts: /workdir/build/zephyr/zephyr.dts
-- Generated pickled edt: /workdir/build/zephyr/edt.pickle
-- Generated zephyr.dts: /workdir/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /workdir/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /workdir/build/zephyr/dts.cmake
CMake Warning at /workdir/zephyr/cmake/modules/dts.cmake:426 (message):
  dtc raised one or more warnings:

  /workdir/build/zephyr/zephyr.dts:46.39-53.5: Warning (interrupt_provider):
  /soc/interrupt-controller@3ff00104: Missing #address-cells in interrupt
  provider

Call Stack (most recent call first):
  /workdir/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /workdir/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /workdir/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

warning: The choice symbol CONSOLE_GETCHAR (defined at subsys/console/Kconfig:16) was selected (set
=y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_CONSOLE_GETCHAR and/or look up
CONSOLE_GETCHAR in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

Parsing /workdir/zephyr/Kconfig
Loaded configuration '/workdir/zephyr/boards/espressif/esp32_devkitc_wroom/esp32_devkitc_wroom_procpu_defconfig'
Merged configuration '/workdir/esp-serial-flasher/examples/zephyr_example/prj.conf'
Configuration saved to '/workdir/build/zephyr/.config'
Kconfig header saved to '/workdir/build/zephyr/include/generated/zephyr/autoconf.h'
[... 79 lines cut ...]
# Zephyr SDK

RUN \
  apt-get -y update \
  && apt-get -y install --no-install-recommends \
  wget \
  xz-utils \
  && apt-get clean \
  && rm -rf /var/lib/apt/lists/* \
  && west sdk install -t xtensa-espressif_esp32_zephyr-elf -H

# ESP Flasher

RUN \
  apt-get -y update \
  && apt-get -y install --no-install-recommends \
  unzip \
  && apt-get clean \
  && rm -rf /var/lib/apt/lists/* \
  && git clone https://github.com/espressif/esp-serial-flasher.git \
  && wget https://dl.espressif.com/esp-at/firmwares/esp32/ESP32-PICO-D4/ESP32-PICO-D4-AT-V3.2.1.0.zip \
  && unzip ESP32-PICO-D4-AT-V3.2.1.0.zip \
  && mv ESP32-PICO-D4-AT-V3.2.1.0/ESP32-PICO-D4-AT-V3.2.1.0/esp-at.bin esp-serial-flasher/examples/binaries/ESP32_AT_Firmware/
# && west build -p -b esp32_devkitc_wroom esp-serial-flasher/examples/zephyr_example/
```

beriberikix, 2024-11-22:
Adding the `include` & `port` directories fixed the CMake issues:

```
zephyr_library_include_directories(
  ../common
  ../binaries
  ../../include
  ../../port
  )
```

But I'm getting linking issues:

```
[197/202] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /workdir/build/zephyr/zephyr_pre0.map 
: && /root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc  -gdwarf-4 -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/workdir/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/workdir/build/zephyr  zephyr/arch/common/libisr_tables.a  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -L"/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/xtensa-espressif_esp32_zephyr-elf"/lib/.  -lgcc  -T/workdir/modules/hal/espressif/zephyr/esp32/src/linker/esp32.rom.alias.ld  -T/workdir/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.ld  -T/workdir/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.api.ld  -T/workdir/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -T/workdir/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld  -T/workdir/modules/hal/espressif/zephyr/esp32/../../components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld  -T/workdir/modules/hal/espressif/zephyr/esp32/../../components/soc/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=longjmp  -lc -L"/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0" -lm -lc -lgcc -lc && cd /workdir/build/zephyr && /usr/bin/cmake -E true
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: app/libapp.a(example_common.c.obj):(.literal.connect_to_target+0x28): undefined reference to `esp_loader_connect'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: app/libapp.a(example_common.c.obj):(.literal.connect_to_target+0x2c): undefined reference to `esp_loader_get_target'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: app/libapp.a(example_common.c.obj):(.literal.connect_to_target+0x30): undefined reference to `esp_loader_change_transmission_rate'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: app/libapp.a(example_common.c.obj):(.literal.connect_to_target+0x34): undefined reference to `loader_port_change_transmission_rate'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: app/libapp.a(example_common.c.obj):(.literal.flash_binary+0x24): undefined reference to `esp_loader_flash_start'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: app/libapp.a(example_common.c.obj):(.literal.flash_binary+0x28): undefined reference to `esp_loader_flash_write'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: app/libapp.a(example_common.c.obj): in function `connect_to_target':
/workdir/esp-serial-flasher/examples/common/example_common.c:263: undefined reference to `esp_loader_connect'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /workdir/esp-serial-flasher/examples/common/example_common.c:277: undefined reference to `esp_loader_get_target'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /workdir/esp-serial-flasher/examples/common/example_common.c:280: undefined reference to `esp_loader_change_transmission_rate'
[... cut ...] efined reference to `esp_loader_get_target'
/root/zephyr-sdk-0.17.0/xtensa-espressif_esp32_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32_zephyr-elf/bin/ld.bfd: /workdir/esp-serial-flasher/examples/zephyr_example/src/main.c:73: undefined reference to `esp_loader_reset_target'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /workdir/build
```

DNedic, 2024-11-22:
Hi @beriberikix thank you for your detailed issue report! At first glance, it seems like you're not using the library as a Zephyr module. You can take a look at the [zephyr](https://github.com/espressif/esp-serial-flasher/tree/master/zephyr) folder for more context.

If you want to use it as a third party CMake library, you will have to do the legwork found there yourself, just like you would for any user port (though admittedly it's quite easy).

Generally, you shouldn't be adding those includes in the example itself, the example should only depend on `esp-serial-flasher` and the example common bits.

beriberikix, 2024-11-22:
Thank you for the reply! I was just testing out the example - it appeared like it should work standalone. Is that not the case?

DNedic, 2024-11-22:
PTAL at how we build it in [our CI](https://github.com/espressif/esp-serial-flasher/blob/master/.gitlab-ci.yml#L226), we use `esp-serial-flasher` as a module.

beriberikix, 2024-11-22:
Ah, the extra module was the missing piece! Should that be added to the [README](https://github.com/espressif/esp-serial-flasher/tree/master/examples/zephyr_example#build-and-flash)?

In any case, here's the updated Dockerfile for future reference:

```dockerfile
FROM debian:stable-slim

# OS dependencies and packages

RUN \
  apt-get -y update \
  && apt-get -y install --no-install-recommends \
  ca-certificates \
  cmake \
  device-tree-compiler \
  git \
  ninja-build \
  && apt-get clean \
  && rm -rf /var/lib/apt/lists/*

# Python

ENV VIRTUAL_ENV=/opt/venv
RUN \
  apt-get -y update \
  && apt-get -y install --no-install-recommends \
  python3 \
  python3-pip \
  python3-venv \
  && python3 -m venv $VIRTUAL_ENV
ENV PATH="$VIRTUAL_ENV/bin:$PATH"

# West

RUN pip install --no-cache-dir wheel west

WORKDIR /workdir

# Zephyr setup

RUN \
  west init \
  && west update --narrow -o=--depth=1 \
  && west zephyr-export \
  && pip install -r zephyr/scripts/requirements.txt

# Zephyr SDK

RUN \
  apt-get -y update \
  && apt-get -y install --no-install-recommends \
  wget \
  xz-utils \
  && apt-get clean \
  && rm -rf /var/lib/apt/lists/* \
  && west sdk install -t xtensa-espressif_esp32_zephyr-elf -H

# ESP Flasher

RUN \
  apt-get -y update \
  && apt-get -y install --no-install-recommends \
  unzip \
  && apt-get clean \
  && rm -rf /var/lib/apt/lists/* \
  && git clone https://github.com/espressif/esp-serial-flasher.git \
  && wget https://dl.espressif.com/esp-at/firmwares/esp32/ESP32-PICO-D4/ESP32-PICO-D4-AT-V3.2.1.0.zip \
  && unzip ESP32-PICO-D4-AT-V3.2.1.0.zip \
  && mv ESP32-PICO-D4-AT-V3.2.1.0/ESP32-PICO-D4-AT-V3.2.1.0/esp-at.bin esp-serial-flasher/examples/binaries/ESP32_AT_Firmware/ \
  && west build -p -b esp32_devkitc_wroom esp-serial-flasher/examples/zephyr_example/ -DZEPHYR_EXTRA_MODULES=/workdir/esp-serial-flasher
```

closed 2024-11-22 by beriberikix

DNedic, 2024-11-22:
You're absolutely correct, we will add this to the README, and thanks for providing this Dockerfile!