espressif/esp-serial-flasher #125: There is always an error burning C2.

vvvyky, 2024-11-30:
### Port

ESP32

### Target chip

ESP32C2

### Hardware Configuration

ESP32-C2 4MB FLASH 26MHZ

### Log output

```plain
Initializing serial port... 115200

Chip model: ESP32-C2

Flash size: 0MB

MAC: d0:ef:76:9c:7a:14

Target chip: ESP32-C2

Ecosystem version: 1

Secure Boot: Disabled

Flash Encryption: Disabled

Secure Download Mode: Disabled

Secure Boot Key 0 Disabled: Disabled

Secure Boot Key 1 Disabled: Disabled

Secure Boot Key 2 Disabled: Disabled

JTAG access: Enabled

USB access: Enabled

Data cache in UART download mode: Enabled

Instruction cache in UART download mode: Enabled

Connection successful!

Erasing flash, programming...

Flash erase failed!
```

### More Information

There is always an error in burning C2. I can't identify the storage. What is the reason?

DNedic, 2024-11-30:
Interesting, it looks like the flash either isn't getting detected or isn't there at all. Can you for a start check if your board/module has the flash chip onboard? We've had cases where some third party boards didn't have flash installed at all.

Dzarda7, 2024-11-30:
Hi @vvvyky, thanks for noticing this. Can you please try to flash the chip using the [esp32_example](https://github.com/espressif/esp-serial-flasher/tree/master/examples/esp32_example) and paste the log here? It Is possible that flash detection registers Are incorrect.

vvvyky, 2024-12-01:
> 你好@vvvyky，感谢您注意到这一点。你能试着用闪存芯片吗[esp32 _示例](https://github.com/espressif/esp-serial-flasher/tree/master/examples/esp32_example)然后把日志粘贴到这里？flash检测寄存器可能不正确。

It is the modification of this esp32_example routine, except that esp_loader_flash_detect_size is added. The detection chip fails to detect the size of the flash memory in C2 chip in detail, and the recording fails at the same time. The parameters of 32, s2, s3 and c3 are fine.

vvvyky, 2024-12-01:
> Hi @vvvyky, thanks for noticing this. Can you please try to flash the chip using the [esp32_example](https://github.com/espressif/esp-serial-flasher/tree/master/examples/esp32_example) and paste the log here? It Is possible that flash detection registers are incorrect.

It is the modification of this esp32_example routine, except that esp_loader_flash_detect_size is added. The detection chip fails to detect the size of the flash memory in C2 chip in detail, and the recording fails at the same time. The parameters of 32, s2, s3 and c3 are fine.

vvvyky, 2024-12-01:
> 有趣的是，看起来这种闪光要么没有被探测到，要么根本就不存在。你能先检查一下你的主板/模块上是否有闪存芯片吗？我们遇到过一些第三方主板根本没有安装flash的情况。

C2 is an on-chip flash memory, and I don't know why the size can't be detected.

vvvyky, 2024-12-01:
> Interesting, it looks like the flash either isn't getting detected or isn't there at all. Can you for a start check if your board/module has the flash chip onboard? We've had cases where some third party boards didn't have flash installed at all.

C2 is an on-chip flash memory, and I don't know why the size can't be detected.

DNedic, 2024-12-13:
@vvvyky could you please enable serial tracing as described in the issue template and submit the logs?

dobairoland, 2025-01-06:
I'm closing this for inactivity. Please let us know if this is still a valid issue.

vvvyky, 2025-03-13:
> Hi [@vvvyky](https://github.com/vvvyky), thanks for reporting this, I was able to reproduce this issue and fixed it appropriately [here](https://github.com/espressif/esp-serial-flasher/commit/ed43e0ac763777ee002c424ae3d0a4c858abdf8b). There was a problem with flash SPI pins reading. You can try it again.

ESP8684H2 can detect successful flash burning, but ESP8684H4 still cannot detect flash burning failure

Dzarda7, 2025-03-17:
@vvvyky thanks and sorry, there are multiple revisions of C2 and there seems to be different SPI conf for each of them. I will investigate it ASAP.

vvvyky, 2025-08-28:
> [@vvvyky](https://github.com/vvvyky) 感谢并抱歉，C2有多个版本，似乎每个版本的SPI配置都不同。我将尽快调查此事。

Still unable to burn ESP32c2, are there any further research questions?

Dzarda7, 2025-09-01:
Hello @vvvyky, now it should be fixed properly on all revisions, sorry for the delay.

vvvyky, 2025-09-06:
> 你好 [@vvvyky](https://github.com/vvvyky)，现在所有版本的问题都应该已经得到妥善解决了。对于造成的延迟，我们深表歉意。

Thank you, there were no issues with the test. Perfect repair

Dzarda7, 2025-09-06:
Sorry it took so long and thanks for the report.