espressif/esp-serial-flasher #127: Creating a standalone Zephyr example

beriberikix, 2024-12-21:
### Port

Zephyr

### Target chip

ESP32

### Hardware Configuration

Currently testing with the host as a nRF52840 DK and an ESP32. 

| nRF | ESP |
|--------|--------|
| p1.04 (RX) | TX |
| p1.05 (TX) | RX |
| p0.16 | Boot |
| p0.24 | En |
| gnd | gnd | 

I've tested with a ESP32-DevKitC but also tested with a ESP32-C6-DevKitC to see if my results are different (they're not.)

### Log output

```plain
I don't think SERIAL_FLASHER_DEBUG_TRACE is working on the Zephyr port or I'm doing something wrong. I enabled it with `menuconfig` and didn't see a change in output.

After `west flash` on the nRF DK, I see:

*** Booting Zephyr OS build v4.0.0 ***
Running ESP Flasher from Zephyr
Cannot connect to target. Error: TIMEOUT
Check if the host and the target are properly connected.
```

On the ESP32 or ESPC6:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0/SDIO_FEI_FEO))
waiting for download
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd5c
load:0x4086e610,len:0x2b20
load:0x40875720,len:0x17e8
entry 0x4086c410
I (23) boot: ESP-IDF v5.1-dev-4557-gf404fe96b1 2nd stage bootloader
I (24) boot: compile time May 20 2023 17:09:13
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
W (48) bootloader_random: bootloader_random_enable() has not been implemented yet
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot:  3 ot_storage       Unknown data     01 3a 00110000 00002000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=42030020 size=0cee8h ( 52968) map
I (120) esp_image: segment 1: paddr=0001cf10 vaddr=40800000 size=03108h ( 12552) load
I (124) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2ac08h (175112) map
I (162) esp_image: segment 3: paddr=0004ac30 vaddr=40803108 size=0abe0h ( 44000) load
I (173) esp_image: segment 4: paddr=00055818 vaddr=4080dcf0 size=018b0h (  6320) load
I (178) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
W (182) bootloader_random: bootloader_random_enable() has not been implemented yet
I (201) cpu_start: Unicore app
I (202) cpu_start: Pro cpu up.
W (210) clk: esp_perip_clk_init() has not been implemented yet
I (217) cpu_start: Pro cpu start user code
I (217) cpu_start: cpu freq: 160000000 Hz
I (217) cpu_start: Application information:
I (220) cpu_start: Project name:     esp_ot_rcp
I (225) cpu_start: App version:      1
I (230) cpu_start: Compile time:     May 20 2023 17:08:16
I (236) cpu_start: ELF file SHA256:  f55f29f7cd1c2393...
I (242) cpu_start: ESP-IDF:          v5.1-dev-4557-gf404fe96b1
I (248) cpu_start: Min chip rev:     v0.0
I (253) cpu_start: Max chip rev:     v0.99 
I (258) cpu_start: Chip rev:         v0.0
I (262) heap_init: Initializing. RAM available for dynamic allocation:
I (270) heap_init: At 40812C40 len 000699D0 (422 KiB): D/IRAM
I (276) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (283) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (290) spi_flash: detected chip: generic
I (294) spi_flash: flash io: dio
W (298) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (311) sleep: Configure to isolate all GPIO pins in sleep state
I (318) sleep: Enable automatic switching of GPIO sleep configuration
I (325) coexist: coex firmware version: ebddf30
I (330) coexist: coexist rom version 5b8dcfa
I (335) app_start: Starting scheduler on CPU0
I (340) main_task: Started on CPU0
�(Jޡ;b�'����F��(�{B�,B;��
```
```

### More Information

I created a [standalone application](https://github.com/beriberikix/pargonesp/tree/devel) that's essentially the example, with a `west.yml` and modified CMake paths. It's likely I made errors when copy files around but couldn't find anything obvious. As you can see from the output, the device does go into programming mode. I tried increasing timeouts but it didn't have any effect.

DNedic, 2024-12-22:
Hello @beriberikix , it does indeed seem that we are not handling `SERIAL_FLASHER_DEBUG_TRACE` on the `Zephyr` port, we will look into this first so that you can submit the logs.

Additionally, you can try debugging the host in order to find the point where the timeout happens and report that as well.

Dzarda7, 2025-01-02:
Hi @beriberikix, thanks for using this library and sorry it takes so long to fix the debug tracing. I already tested the fix and it works, but I need to fix some non-related issues first. Please can you add the following lines into `zephyr/CMakeLists.txt` of the library? This should fix the impossibility to enable debug trace and you can try to debug it a bit easier or post the log here.

```
if(DEFINED SERIAL_FLASHER_DEBUG_TRACE OR CONFIG_SERIAL_FLASHER_DEBUG_TRACE)
    target_compile_definitions(esp_flasher INTERFACE SERIAL_FLASHER_DEBUG_TRACE=1)
 endif()
```

beriberikix, 2025-01-03:
No worries, thank you! In addition to the CMake edit, I also need to add `CONFIG_SERIAL_FLASHER_DEBUG_TRACE=y`.

I now see more output from the terminal but it doesn't seem to be all that helpful.

```
*** Booting Zephyr OS build v4.0.0 ***
Running ESP Flasher from Zephyr

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 
--- READ ---
80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 Cannot connect to target. Error: TIMEOUT
Check if the host and the target are properly connected.
```

Dzarda7, 2025-01-03:
Can you please review that the pinout is correct? I just noticed that RX and TX pins are swapped in the README for Zephyr example which might have confused you. The example works for me after that.

Also beware that the target (ESP32) should not be connected to a PC through USB during flashing as USB-UART converter might interrupt the transfer, just power the target through pins. I see you tried to read the log from the target as you attached it (the flashing should not work like this, as I said). From the `waiting for download` line it seems that the target is able to get into the download mode if the timing is correct.

beriberikix, 2025-01-03:
I tried powering the ESP32C6 direct while programming and got the same result.

```
*** Booting Zephyr OS build v4.0.0 ***
Running ESP Flasher from Zephyr

--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0
--- READ ---
45 53 50 2d 52 4f 4d 3a 65 73 70 33 32 63 36 2d 32 30 32 32 30 39 31 39 0d 0a 42 75 69 6c 64 3a 53 65 70 20 31 39 20 32 30 32 32 0d 0a 72 73 74 3a 30 78 31 20 28 50 4f 57 45 52 4f 4e 29 2c 62 6f 6f 74 3a 30 78 34 20 28 44 4f 57 4e 4c 4f 41 44 28 55 53 42 2f 55 41 52 54 30 2f 53 44 49 4f 5f 46 45 49 5f 46 45 4f 29 29 0d 0a 77 61 69 74 69 6e 67 20 66 6f 72 20 64 6f 77 6e 6c 6f 61 64 0d 0a
--- WRITE ---
c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 c0 00 08 24 00 00 00 00 00 07 07 12 20 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c0 Cannot connect to target. Error: TIMEOUT
Check if the host and the target are properly connected.
```

I also confirmed the hookup diagrams are as a described above and defined in my [overlay](https://github.com/beriberikix/pargonesp/blob/devel/boards/nrf52840dk_nrf52840.overlay).

Dzarda7, 2025-01-03:
I tried your code from the [standalone application](https://github.com/beriberikix/pargonesp/tree/devel) with esp32 as a host and esp32 (and esp32c6) as a target and it works for me, so it seems like some issue with the nRF chip, overlay or a hardware issue. Also please be aware that GPIO0 is not the boot pin on esp32c6, it is GPIO9. Just making sure once again, USB connector of the target should be disconnected.

beriberikix, 2025-01-04:
Appreciate the testing! I've confirmed that hardware isn't likely issue on both sides, so it's possible it's the overlay. Can you share the overlay you used for the host ESP32?

Dzarda7, 2025-01-04:
It can be found [here](https://github.com/espressif/esp-serial-flasher/blob/master/examples/zephyr_example/socs/esp32_procpu.overlay).

beriberikix, 2025-01-04:
In case there was a discrepancy between how I thought I wired the boards and what I actually did, I threw together this diagram.
[image]

Dzarda7, 2025-01-04:
One thing I noticed, the pins you selected for UART are marked as low frequency I/O only which should be used below 10 KHz as stated [here](https://docs.zephyrproject.org/latest/boards/nordic/nrf52840dk/doc/index.html#selecting-the-pins), you can try to change the pins.

closed 2025-01-23 by Dzarda7